google-coral/libedgetpu #1: About the libedgetpu porting to MCU

akioolin, 2020-06-09:
Dear Sir:

I'm very glade to see the libedgetpu had been open sourced. for the 
current release, it should be Linux based. If I want to port this library to 
MCU based solution, which part should I take care?

Like NVDLA's KMD, it divides into two part. one is linux porting interface,
the other is firmware, which is platform independent. is current libedgetpu
design has this layer to take about porting?

Thank you very much.

Best Regards,
Akio

usbguru, 2020-10-27:
Hi Akio,
If you want to get TFlite/MCU models running on the edgetpu, the libedgetpu code is a useful guide for understanding the process, but not ideal for porting.  It has too much code that won’t be needed, and it would be hard to maintain libedgetpu if it targeted multiple physical layers, multiple OS, and RTOS, and multiple ML frameworks.
 
You can run unaccelerated TFlite models on an MCU today with Tensorflow Lite/MCU:   https://www.tensorflow.org/lite/microcontrollers
 
Tensorflow includes a delegate mechanism that allows some or most of the model computations to be passed to a GPU or TPU:  https://www.tensorflow.org/lite/performance/delegates
 
For the edgetpu this is implemented as a CustomOp that TFlite passes to the edgetpu:  https://www.tensorflow.org/lite/guide/ops_custom
 
A suitable MCU delegate is actively being worked on right now.

closed 2020-10-27 by usbguru

ilouzl, 2021-07-06:
Any update on the MCU delegate development?
Maybe someone could point out what parts of the libedgetpu would be relevant for such port?