google-coral/libedgetpu #6: Build failed on Windows 10

NobuoTsukamoto, 2020-11-21:
### System information
- Windows 10 Home 20H2 Build 19042.630 64bit
- Visual Studio Community 2019 Version 16.8.2
- Bazel 3.4.1

### Describe the current behavior
When I build libedgetpu (Release Frogfish) on Windows 10 it fails with the following error:
x64 Native Tools Command Prompt for VS 2019
```
>build.bat
...
.\driver/kernel/windows/windows_gasket_ioctl.inc(112): error C3861: 'StringPrintf': identifier not found
.\driver/kernel/windows/windows_gasket_ioctl.inc(120): error C3861: 'StringPrintf': identifier not found
.\driver/kernel/windows/windows_gasket_ioctl.inc(128): error C3861: 'StringPrintf': identifier not found
.\driver/kernel/windows/windows_gasket_ioctl.inc(155): error C3861: 'StringPrintf': identifier not found
...
```

I think that ".\driver/kernel/windows/windows_gasket_ioctl.inc" is missing a header file include.
```
#include "port/stringprintf.h"
```

Namburger, 2020-11-23:
@NobuoTsukamoto Thanks for reporting!

dmitriykovalev, 2021-07-28:
I believe this is already fixed, please try the latest source code from Grouper release.

closed 2021-07-28 by dmitriykovalev