google-coral/libedgetpu #9: Possible deadlock

YijinLiu, 2020-12-16:
My program is experiencing deadlock in libedgetpu occasionally. Following is the stack trace.

<pre>
Thread 58 (Thread 0x7fff7d7fa700 (LWP 24554)):
#0  0x00007ffff75ebad3 in futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x7fff7d7f9170) at ../sysdeps/unix/sysv/linux/futex-internal.h:88
#1  __pthread_cond_wait_common (abstime=0x0, mutex=0x7fff7d7f9120, cond=0x7fff7d7f9148) at pthread_cond_wait.c:502
#2  __pthread_cond_wait (cond=0x7fff7d7f9148, mutex=0x7fff7d7f9120) at pthread_cond_wait.c:655
#3  0x00007ffff78b48bc in std::condition_variable::wait(std::unique_lock<std::mutex>&) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007fffea8f9a0e in std::condition_variable::wait<platforms::darwinn::BlockingCounter::Wait()::<lambda()> >(std::unique_lock<std::mutex> &, platforms::darwinn::BlockingCounter::<lambda()>) (this=0x7fff7d7f9148, 
    __lock=..., __p=...) at /usr/include/c++/5/condition_variable:98
#5  0x00007fffea8f99c4 in platforms::darwinn::BlockingCounter::Wait (this=0x7fff7d7f9120) at port/blocking_counter.cc:42
#6  0x00007fffea8ce763 in platforms::darwinn::driver::Driver::Execute (this=0xc1a6700, request=std::shared_ptr<platforms::darwinn::api::Request> (empty) = {...}) at driver/driver.cc:593
#7  0x00007fffea8b9f0a in platforms::darwinn::tflite::EdgeTpuDriverWrapper::InvokeExecutable (this=0xbdf57a0, context=0xc087628, node=0xc13f088) at tflite/edgetpu_context_direct.cc:240
#8  0x00007fffea8ac653 in platforms::darwinn::tflite::(anonymous namespace)::CustomOpInvoke (context=0xc087628, node=0xc13f088) at tflite/custom_op_direct.cc:122
#9  0x00000000068df998 in tflite::impl::Subgraph::Invoke() ()
#10 0x0000000006830e87 in tflite::impl::Interpreter::Invoke() ()

Thread 32 (Thread 0x7fffb4cf5700 (LWP 24515)):
#0  0x00007ffff75ebad3 in futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0xc1a6a14) at ../sysdeps/unix/sysv/linux/futex-internal.h:88
#1  __pthread_cond_wait_common (abstime=0x0, mutex=0xc19a4b0, cond=0xc1a69e8) at pthread_cond_wait.c:502
#2  __pthread_cond_wait (cond=0xc1a69e8, mutex=0xc19a4b0) at pthread_cond_wait.c:655
#3  0x00007ffff78b48bc in std::condition_variable::wait(std::unique_lock<std::mutex>&) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007fffea87517f in std::_V2::condition_variable_any::wait<platforms::darwinn::driver::(anonymous namespace)::Lock2> (this=0xc1a69e8, __lock=...) at /usr/include/c++/5/condition_variable:250
#5  0x00007fffea871789 in platforms::darwinn::driver::UsbDriver::WorkerThreadFunc (this=0xc1a6700) at driver/usb/usb_driver.cc:1322
#6  0x00007fffea8728c7 in platforms::darwinn::driver::UsbDriver::<lambda()>::operator()(void) const (__closure=0xc1cb358) at driver/usb/usb_driver.cc:1604
#7  0x00007fffea878834 in std::_Bind_simple<platforms::darwinn::driver::UsbDriver::DoOpen(bool)::<lambda()>()>::_M_invoke<>(std::_Index_tuple<>) (this=0xc1cb358) at /usr/include/c++/5/functional:1531
#8  0x00007fffea8787b5 in std::_Bind_simple<platforms::darwinn::driver::UsbDriver::DoOpen(bool)::<lambda()>()>::operator()(void) (this=0xc1cb358) at /usr/include/c++/5/functional:1520
#9  0x00007fffea87875c in std::thread::_Impl<std::_Bind_simple<platforms::darwinn::driver::UsbDriver::DoOpen(bool)::<lambda()>()> >::_M_run(void) (this=0xc1cb340) at /usr/include/c++/5/thread:115
#10 0x00007ffff78baa50 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#11 0x00007ffff75e56db in start_thread (arg=0x7fffb4cf5700) at pthread_create.c:463
#12 0x00007fffea0d171f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
</pre>

I use one thread to run the inference. This only happens about once per week. I haven't find a way to repro it so far.

Namburger, 2020-12-16:
Hello, can you describe what exactly happens and how is your program designed?

YijinLiu, 2020-12-16:
We run a SSD model using edgetpu. A single thread receives images and run the model.
Once a while, the thread stuck forever. The initial post has the stuck trace. Unfortunately, I have no luck finding a way to repro so far. Let me know if you need more information.

Namburger, 2020-12-21:
@YijinLiu I see, may I ask if this is over usb3 or usb2?

YijinLiu, 2020-12-22:
It's usb3. Thanks.

manoj7410, 2021-07-05:
Feel free to reopen if this is still an issue.

YijinLiu, 2021-10-12:
Well, even more weird: I am not allowed to reopen it!

manoj7410, 2021-10-12:
@YijinLiu Issue reopened. Can you please share any steps to reproduce ? 
Which platform are you working on ?

YijinLiu, 2021-10-12:
It's a desktop machine, with CPU i7-10700, running ubuntu 20, with kernel 5.11.
I only have one USB accelerator connected to the machine.

We have users with many other configurations that are experiencing the same issue.

YijinLiu, 2021-11-09:
There is another issue, a likely contention:
- The worker thread (from driver/usb/usb_driver.cc) installs these event/interrupt/bulk-in readers, which timeout every 6 seconds by default.
- The client calls UsbDriver::DoSubmit to submit jobs to the device.
- If the readers timeout the same time UsbDriver::ProcessIo issues bulk-out requests, it may be stuck there for ever.
I changed usb_timeout_millis to 0 to disable the timeout. It has been running for 10 days without the deadlock I saw before.

manoj7410, 2021-11-29:
Thank you for sharing these details. What all files were modified to solve the issue ?

chao-camect, 2021-11-29:
Besides the diffs I posted previously, here are two extra changes:

<pre>
diff --git a/driver/beagle/beagle_usb_driver_provider.cc b/driver/beagle/beagle_usb_driver_provider.cc
index dcfd804..2b82a00 100644
--- a/driver/beagle/beagle_usb_driver_provider.cc
+++ b/driver/beagle/beagle_usb_driver_provider.cc
@@ -105,7 +105,7 @@ ABSL_FLAG(bool, usb_enable_bulk_descriptors_from_device,
 ABSL_FLAG(bool, usb_enable_processing_of_hints,
           GetEnv("USB_ENABLE_PROCESSING_OF_HINTS", true),
           "USB set to true for driver to proactively send data to device.");
-ABSL_FLAG(int, usb_timeout_millis, GetEnv("USB_TIMEOUT_MILLIS", 6000),
+ABSL_FLAG(int, usb_timeout_millis, GetEnv("USB_TIMEOUT_MILLIS", 0),
           "USB timeout in milliseconds");
 ABSL_FLAG(bool, usb_reset_back_to_dfu_mode,
           GetEnv("USB_RESET_BACK_TO_DFU_MODE", false),
@@ -135,7 +135,7 @@ ABSL_FLAG(bool, usb_enable_overlapping_requests,
           "the current one.");
 ABSL_FLAG(bool, usb_enable_overlapping_bulk_in_and_out,
           GetEnv("USB_ENABLE_OVERLAPPING_BULK_IN_AND_OUT", true),
-          "Allows bulk-in trasnfer to be submitted before previous bulk-out "
+          "Allows bulk-in transfer to be submitted before previous bulk-out "
           "requests complete.");
 ABSL_FLAG(bool, usb_enable_queued_bulk_in_requests,
           GetEnv("USB_ENABLE_QUEUED_BULK_IN_REQUESTS", true),
diff --git a/driver/usb/usb_driver.h b/driver/usb/usb_driver.h
index 182f27e..d3580a9 100644
--- a/driver/usb/usb_driver.h
+++ b/driver/usb/usb_driver.h
@@ -151,7 +151,7 @@ class UsbDriver : public Driver {
     bool usb_fail_if_slower_than_superspeed{false};
 
     // General timeout for USB operations in milliseconds.
-    int usb_timeout_millis{6000};
+    int usb_timeout_millis{0};
 
     // If non-empty, the firmware image to use for automatic DFU.
     // This feature is only available when a device factory has been supplied.
</pre>

manoj7410, 2021-11-30:
@YijinLiu Are these changes made to the generic Linux drivers?

chao-camect, 2021-11-30:
It's libedgetpu. You can find the filenames in the diff...

manoj7410, 2021-12-02:
@chao-camect Got it. We are still not able to reproduce this. However, can you create a patch of the complete code change and submit for review?

chao-camect, 2021-12-02:
To repro it, you'll need to add random sleep time between two calls to USB Coral. Since the default timeout is 6s, the sleep time need to be around that to replicate.
I'll find some time to create the PR.

xxs1989, 2022-09-28:
Hey, I got the same issue as :

hjonnala, 2022-09-29:
Hi @xxs1989 the board requires 5V/2A power supply. Please check the section 5.1 for more details at: https://coral.ai/static/files/Coral-Dev-Board-Mini-datasheet.pdf. Thanks!

messani, 2024-10-19:
Hi.

I have exactly same experience. I have Raspberry pi 4 and raspberry pi. I am using CodeProject.AI in docker. Both raspberries have problems with USB TPU Coral. Both raspberries were in deadlocked state after few hours (or approx. 10k requests). I tried to use USB hub and Y cable. No success.

So I tried to compile libedgetpu with your patches. First patch (order of locks) had no success. But after that I applied second patch (change usb_timeout_millis from 6000 to 0) it seems to be ok now.

Now uptime is almost two days and there are more than 600k requests. 

If you need some additional information, I can you do some research for you. I am able to build debug version and I am able to attach it by GDB and send you some info. If you would like to add extra logging, I can do it for you too.

I am developer (C/C++/CSharp), but I do not understand USB data.