google-coral/libedgetpu #11: Failed to set performance expectation for M.2 edgetpu

YijinLiu, 2021-01-23:
The error message shows:
<pre>
W :131] Could not set performance expectation : 11 (Inappropriate ioctl for device)
</pre>
I am using Ubuntu 18.04, with kernel 5.4.0. I don't see this error message for USB accelerators. 
I get 7ms/frame on a [USB accelerator](https://coral.ai/products/accelerator/), only 15ms on the [M.2 one](https://coral.ai/products/m2-accelerator-dual-edgetpu). Could the speed difference caused by the error?

marcin-kamionowski, 2021-02-10:
Hi!
I also have problem like this. I'm using apex driver that comes from mainline linux. It does not support this [ioctl](https://github.com/torvalds/linux/blob/master/drivers/staging/gasket/apex.h). 

I tried https://coral.googlesource.com/linux-imx/+/refs/heads/dkms/drivers/staging/gasket/ but I get some errors on device initialization. Not sure why.

mb-karo, 2021-04-07:
Hello!

I am also receiving this message with NXP linux-imx kernel 5.4.70 which is based on Mainline Kernel.

Is this a real "problem" or just a message? My inference is running fine.

Namburger, 2021-04-07:
Hey guys, I suggest blacklisting the mainline gasket module and use the one we provided

closed 2021-07-01 by manoj7410

soldierkam, 2021-11-16:
Most likely it still uses old version. Please make sure that kernel loaded newer version of modules. There are two of them - gasket and apex (or smth similar).

hack2spider, 2021-11-16:
> 
> 
> Most likely it still uses old version. Please make sure that kernel loaded newer version of modules. There are two of them - gasket and apex (or smth similar).

thanks for reply .. 

I`ve installed the drivers as follow : 

1,
echo “deb https://packages.cloud.google.com/apt coral-edgetpu-stable main” | tee /etc/apt/sources.list.d/coral-edgetpu.list

curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | apt-key add -

apt install gasket-dkms

sh -c “echo ‘SUBSYSTEM==“apex”, MODE=“0660”, GROUP=“apex”’ >> /etc/udev/rules.d/65-apex.rules”

groupadd apex

adduser $USER apex

apt-get update

apt-get install libedgetpu1-std

now my question is: 

was I supposed to do these on the turnkey lxc ? or only on pve ?