google-coral/libedgetpu #13: Fail to compile  libedgetpu

aquaXU, 2021-02-22:
Dear all,

I got an error when I compiled  libedgetpu libary,

make libedgetpu-direct CPU=aarch64:

`bazel build --sandbox_debug --subcommands --experimental_repo_remote_exec --compilation_mode=opt --define darwinn_portable=1 --action_env PYTHON_BIN_PATH=/usr/bin/python3 --cpu=aarch64 --embed_label='TENSORFLOW_COMMIT=f394a768719a55b5c351ed1ecab2ec6f16f99dd4' --stamp --verbose_failures --crosstool_top=@crosstool//:toolchains --compiler=gcc //tflite/public:libedgetpu_direct_all.so
Starting local Bazel server and connecting to it...
... still trying to connect to local Bazel server after 10 seconds ...
INFO: Analyzed target //tflite/public:libedgetpu_direct_all.so (60 packages loaded, 1021 targets configured).
INFO: Found 1 target...
SUBCOMMAND: # @flatbuffers//src:flatc_library [action 'Compiling src/flatc.cpp [for host]', configuration: 3d467bcde7472f94e8494187ceef88f756b47bbba87d97f671b42962e8ccc7d6, execution platform: @local_config_platform//:host]
(cd /home/firefly/.cache/bazel/_bazel_firefly/5e823d200991dd229ee17d0652942eef/execroot/libedgetpu && \
  exec env - \
    PATH=/home/firefly/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin \
    PWD=/proc/self/cwd \
  /usr/bin/aarch64-linux-gnu-gcc -fPIC -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O3 -DNDEBUG '-D_FORTIFY_SOURCE=2' -ffunction-sections -fdata-sections '-march=armv8-a' -funsafe-math-optimizations -ftree-vectorize '-std=c++11' -MD -MF bazel-out/host/bin/external/flatbuffers/src/_objs/flatc_library/flatc.d '-frandom-seed=bazel-out/host/bin/external/flatbuffers/src/_objs/flatc_library/flatc.o' -iquote external/flatbuffers -iquote bazel-out/host/bin/external/flatbuffers -Ibazel-out/host/bin/external/flatbuffers/src/_virtual_includes/flatc_library -Ibazel-out/host/bin/external/flatbuffers/src/_virtual_includes/flatbuffers -g0 -g0 -no-canonical-prefixes -fno-canonical-system-headers -Wno-builtin-macro-redefined '-D__DATE__="redacted"' '-D__TIMESTAMP__="redacted"' '-D__TIME__="redacted"' -c external/flatbuffers/src/flatc.cpp -o bazel-out/host/bin/external/flatbuffers/src/_objs/flatc_library/flatc.o)
SUBCOMMAND: # @flatbuffers//grpc/src/compiler:python_generator_private [action 'Compiling grpc/src/compiler/python_generator.cc [for host]', configuration: 3d467bcde7472f94e8494187ceef88f756b47bbba87d97f671b42962e8ccc7d6, execution platform: @local_config_platform//:host]
(cd /home/firefly/.cache/bazel/_bazel_firefly/5e823d200991dd229ee17d0652942eef/execroot/libedgetpu && \
  exec env - \
    PATH=/home/firefly/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin \
    PWD=/proc/self/cwd \
  /usr/bin/aarch64-linux-gnu-gcc -fPIC -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O3 -DNDEBUG '-D_FORTIFY_SOURCE=2' -ffunction-sections -fdata-sections '-march=armv8-a' -funsafe-math-optimizations -ftree-vectorize '-std=c++11' -MD -MF bazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_objs/python_generator_private/python_generator.d '-frandom-seed=bazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_objs/python_generator_private/python_generator.o' -iquote external/flatbuffers -iquote bazel-out/host/bin/external/flatbuffers -Ibazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_virtual_includes/python_generator_private -Ibazel-out/host/bin/external/flatbuffers/_virtual_includes/flatbuffers -Ibazel-out/host/bin/external/flatbuffers/src/_virtual_includes/flatbuffers -g0 -g0 -no-canonical-prefixes -fno-canonical-system-headers -Wno-builtin-macro-redefined '-D__DATE__="redacted"' '-D__TIMESTAMP__="redacted"' '-D__TIME__="redacted"' -c external/flatbuffers/grpc/src/compiler/python_generator.cc -o bazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_objs/python_generator_private/python_generator.o)
SUBCOMMAND: # @flatbuffers//grpc/src/compiler:go_generator [action 'Compiling grpc/src/compiler/go_generator.cc [for host]', configuration: 3d467bcde7472f94e8494187ceef88f756b47bbba87d97f671b42962e8ccc7d6, execution platform: @local_config_platform//:host]
(cd /home/firefly/.cache/bazel/_bazel_firefly/5e823d200991dd229ee17d0652942eef/execroot/libedgetpu && \
  exec env - \
    PATH=/home/firefly/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin \
    PWD=/proc/self/cwd \
  /usr/bin/aarch64-linux-gnu-gcc -fPIC -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O3 -DNDEBUG '-D_FORTIFY_SOURCE=2' -ffunction-sections -fdata-sections '-march=armv8-a' -funsafe-math-optimizations -ftree-vectorize '-std=c++11' -MD -MF bazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_objs/go_generator/go_generator.d '-frandom-seed=bazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_objs/go_generator/go_generator.o' -iquote external/flatbuffers -iquote bazel-out/host/bin/external/flatbuffers -Ibazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_virtual_includes/go_generator -Ibazel-out/host/bin/external/flatbuffers/_virtual_includes/flatbuffers -Ibazel-out/host/bin/external/flatbuffers/src/_virtual_includes/flatbuffers -g0 -g0 -no-canonical-prefixes -fno-canonical-system-headers -Wno-builtin-macro-redefined '-D__DATE__="redacted"' '-D__TIMESTAMP__="redacted"' '-D__TIME__="redacted"' -c external/flatbuffers/grpc/src/compiler/go_generator.cc -o bazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_objs/go_generator/go_generator.o)
SUBCOMMAND: # @flatbuffers//grpc/src/compiler:swift_generator [action 'Compiling grpc/src/compiler/swift_generator.cc [for host]', configuration: 3d467bcde7472f94e8494187ceef88f756b47bbba87d97f671b42962e8ccc7d6, execution platform: @local_config_platform//:host]
(cd /home/firefly/.cache/bazel/_bazel_firefly/5e823d200991dd229ee17d0652942eef/execroot/libedgetpu && \
  exec env - \
    PATH=/home/firefly/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin \
    PWD=/proc/self/cwd \
  /usr/bin/aarch64-linux-gnu-gcc -fPIC -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O3 -DNDEBUG '-D_FORTIFY_SOURCE=2' -ffunction-sections -fdata-sections '-march=armv8-a' -funsafe-math-optimizations -ftree-vectorize '-std=c++11' -MD -MF bazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_objs/swift_generator/swift_generator.d '-frandom-seed=bazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_objs/swift_generator/swift_generator.o' -iquote external/flatbuffers -iquote bazel-out/host/bin/external/flatbuffers -Ibazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_virtual_includes/swift_generator -Ibazel-out/host/bin/external/flatbuffers/_virtual_includes/flatbuffers -Ibazel-out/host/bin/external/flatbuffers/src/_virtual_includes/flatbuffers -g0 -g0 -no-canonical-prefixes -fno-canonical-system-headers -Wno-builtin-macro-redefined '-D__DATE__="redacted"' '-D__TIMESTAMP__="redacted"' '-D__TIME__="redacted"' -c external/flatbuffers/grpc/src/compiler/swift_generator.cc -o bazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_objs/swift_generator/swift_generator.o)
SUBCOMMAND: # @flatbuffers//grpc/src/compiler:cpp_generator [action 'Compiling grpc/src/compiler/cpp_generator.cc [for host]', configuration: 3d467bcde7472f94e8494187ceef88f756b47bbba87d97f671b42962e8ccc7d6, execution platform: @local_config_platform//:host]
(cd /home/firefly/.cache/bazel/_bazel_firefly/5e823d200991dd229ee17d0652942eef/execroot/libedgetpu && \
  exec env - \
    PATH=/home/firefly/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin \
    PWD=/proc/self/cwd \
  /usr/bin/aarch64-linux-gnu-gcc -fPIC -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O3 -DNDEBUG '-D_FORTIFY_SOURCE=2' -ffunction-sections -fdata-sections '-march=armv8-a' -funsafe-math-optimizations -ftree-vectorize '-std=c++11' -MD -MF bazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_objs/cpp_generator/cpp_generator.d '-frandom-seed=bazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_objs/cpp_generator/cpp_generator.o' -iquote external/flatbuffers -iquote b [... cut ...]
1613987182.506878594: src/main/tools/linux-sandbox-pid1.cc:428: child exited normally with code 0
1613987182.537172251: src/main/tools/linux-sandbox.cc:233: child exited normally with code 0
Target //tflite/public:libedgetpu_direct_all.so failed to build
INFO: Elapsed time: 89.636s, Critical Path: 16.29s
INFO: 91 processes: 91 internal.
FAILED: Build did NOT complete successfully
Makefile:99: recipe for target 'libedgetpu-direct' failed
make: *** [libedgetpu-direct] Error 1`

Environment:
Architecture: aarch64
OS: Ubuntu 18.04.5 LTS

Namburger, 2021-02-23:
@aquaXU I believe you either have some missing library or mismatching dependency. I suggest using the provided docker image to build, we also use this internally since it's became out of hand to have different system and glic version building binaries for a target.
```
$ DOCKER_CPUS="aarch64" DOCKER_TARGETS=libedgetpu make docker-build
```

manoj7410, 2021-07-05:
Feel free to reopen if this is still an issue.

closed 2021-07-05 by manoj7410