google-coral/libedgetpu #14: Update debian packaging files from v14 to v15

pdecat, 2021-03-02:
Currently available debian packaging files are only v14: https://github.com/google-coral/libedgetpu/blob/14eee1a076aa1af7ec1ae3c752be79ae2604a708/debian/changelog

While debian packages available from https://packages.cloud.google.com/apt/dists/coral-edgetpu-stable/main/binary-amd64/Packages are v15

Could you please update the debian packaging files in this repository?

jan-kiszka, 2021-05-04:
Also, where is the true upstream repo for those debian packages? Here?

I'm also facing troubles pulling the sources of the debian packages, to cross-check the difference. But that debian repo seems to miss them (added `deb-src https://packages.cloud.google.com/apt coral-edgetpu-stable main`).

manoj7410, 2021-07-28:
Can you check if you are now able to get the latest library ?

pdecat, 2021-07-28:
Debian packaging files for all releases seem ok now, thanks!

closed 2021-07-28 by manoj7410