google-coral/libedgetpu #16: Anyone running on openSUSE

Spanni26, 2021-04-12:
Does anyone get that stuff running on opensuse?
We use the m2-board.

Regards

Daniel

manoj7410, 2021-05-26:
Is this a duplicate of https://github.com/google-coral/libedgetpu/issues/20 ? @Spanni26

Spanni26, 2021-05-26:
Nope. Same error message but at an other point of the build.

Bazel is a hell of ......

Spanni26, 2021-05-27:
Ok, Cause seems to be the same as in #20 . But how to fix it?

manoj7410, 2021-05-28:
The only solution is to use the docker image to build, we also use this since it is really challenging to have different system and glibc version building binaries for a target.

Spanni26, 2021-05-28:
Hi,

well, we build this on a build server (open build system) to build
packages for different systems at one run. This system does not have an
internet connection, so docker can't be used I think.

Why the hack can't they simply use a "normal" build system instead of
bazel, which (besides of google) nobody is using....

Spanni26, 2021-06-07:
Can I influnce the search path for headers includes? 
I get "src/coral_wrapper.cc:15:10: fatal error: numpy/arrayobject.h: No such file or directory" cause the numpy headers are not under "/usr/include".....

dmitriykovalev, 2021-07-29:
I believe https://github.com/google-coral/libedgetpu/commit/0f1def1f0e0e5ee79a517f30733f09dda2441304 fixes the "include" problem, please try the latest source code from `master` branch.

manoj7410, 2021-08-03:
@Spanni26 Feel free to reopen this thread if the issue still persists.

closed 2021-08-03 by manoj7410