google-coral/libedgetpu #20: missing dependency declarations from flatbuffers

Spanni26, 2021-05-26:
I try to build pycoral on opensuse. This includes building libedgetpu, which uses "flatbuffers". But flatbuffers can't be build cause of a "missing dependency declarations" Error.
I'm not experienced with bazel, but it seems that tools_cc detects wrong include paths.

`[ 1057s] SUBCOMMAND: # @flatbuffers//src:flatc [action 'Compiling external/flatbuffers/src/flatc_main.cpp [for host]', configuration: 4fe28f528853d1b319ce2a9ad37c73d8048f19dc8f3551d117c5585d08b71fef]
[ 1057s] (cd /home/abuild/.cache/bazel/_bazel_abuild/792793a7452147c3117e0614cb9d414a/execroot/pycoral && \
[ 1057s]   exec env - \
[ 1057s]     PATH=/usr/local/bin:/usr/bin:/bin:/usr/lib/mit/bin:/usr/lib/mit/sbin \
[ 1057s]     PWD=/proc/self/cwd \
[ 1057s]   /usr/bin/gcc -fPIC -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -msse4.2 -g0 -O3 -DNDEBUG '-D_FORTIFY_SOURCE=2' -ffunction-sections -fdata-sections '-std=c++11' -MD -MF bazel-out/host/bin/external/flatbuffers/src/_objs/flatc/flatc_main.d '-frandom-seed=bazel-out/host/bin/external/flatbuffers/src/_objs/flatc/flatc_main.o' -iquote external/flatbuffers -iquote bazel-out/host/bin/external/flatbuffers -Ibazel-out/host/bin/external/flatbuffers/src/_virtual_includes/flatc -Ibazel-out/host/bin/external/flatbuffers/src/_virtual_includes/flatc_library -Ibazel-out/host/bin/external/flatbuffers/src/_virtual_includes/flatbuffers -Ibazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_virtual_includes/cpp_generator -Ibazel-out/host/bin/external/flatbuffers/_virtual_includes/flatbuffers -Ibazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_virtual_includes/go_generator -Ibazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_virtual_includes/java_generator -Ibazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_virtual_includes/python_generator -Ibazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_virtual_includes/python_generator_private -Ibazel-out/host/bin/external/flatbuffers/grpc/src/compiler/_virtual_includes/swift_generator -g0 -g0 -no-canonical-prefixes -fno-canonical-system-headers -Wno-builtin-macro-redefined '-D__DATE__="redacted"' '-D__TIMESTAMP__="redacted"' '-D__TIME__="redacted"' -c external/flatbuffers/src/flatc_main.cpp -o bazel-out/host/bin/external/flatbuffers/src/_objs/flatc/flatc_main.o)
[ 1059s] ERROR: /home/abuild/.cache/bazel/_bazel_abuild/792793a7452147c3117e0614cb9d414a/external/flatbuffers/src/BUILD:40:1: undeclared inclusion(s) in rule '@flatbuffers//src:flatc':
[ 1059s] this rule is missing dependency declarations for the following files included by 'external/flatbuffers/src/flatc_main.cpp':
[ 1059s]   '/usr/lib64/gcc/x86_64-suse-linux/7/include/stddef.h'
[ 1059s]   '/usr/lib64/gcc/x86_64-suse-linux/7/include/stdarg.h'
[ 1059s]   '/usr/lib64/gcc/x86_64-suse-linux/7/include/stdint.h'
[ 1059s] src/main/tools/linux-sandbox.cc:154: linux-sandbox-pid1 has PID 2962`

I can't use docker, cause I build on a build system (open build service) which has no connection to the internet.

Any ideas how to fix this?

manoj7410, 2021-05-26:
> [ 1059s] this rule is missing dependency declarations for the following files included by 'external/flatbuffers/src/flatc_main.cpp':
[ 1059s] '/usr/lib64/gcc/x86_64-suse-linux/7/include/stddef.h'
[ 1059s] '/usr/lib64/gcc/x86_64-suse-linux/7/include/stdarg.h'
[ 1059s] '/usr/lib64/gcc/x86_64-suse-linux/7/include/stdint.h'

Are these files present at the mentioned path ?

Spanni26, 2021-05-26:
Yes, But this files are not configured as dependencies it seems. Thats what the build is claiming about.

Spanni26, 2021-05-27:
Ok, it seems not to be a flatbuffsers issue. Flatbuffers standalone builds fine.

The worst thing on all: nobody  can help you with bazel.

Doom4535, 2021-06-06:
Lol, if it's any comfort, I'm having the same issue on Arch...

Doom4535, 2021-06-06:
Does anyone know how to interpret Bazel error messages?  I can't tell if it is saying this project is missing the dependency declaration or if it is the version of Flatbuffers that is pulled in that is missing the declaration?

pmdaye, 2021-07-26:
Just posted an identical issue on Fedora 34....

dmitriykovalev, 2021-07-31:
I hope this is fixed in https://github.com/google-coral/libedgetpu/commit/0f1def1f0e0e5ee79a517f30733f09dda2441304, please reopen if something is still broken.

closed 2021-07-31 by dmitriykovalev

dig9, 2023-05-30:
This issue is over 2 years old and still not fixed. To solve this issue, I used cross compile, docker and colab, etc. and tried all kinds of methods, but I can't figure it out. It seems that it is not possible in ubuntu 22.04 on raspberry, which I use. As a result of the test, it was possible only on ubuntu 20.04.05 (only server) and debian buster 32bit on the Raspberry Pi platform.
  But was this issue closed? It's exactly the same result.