google-coral/libedgetpu #21: Ubuntu 21.04 on Raspberry Pi 4

BernardinD, 2021-05-28:
I've been unable to run the edgetpu lib on my RPi running Ubuntu 21.04. It is an ARMv8 aach64 OS which should be compatible but when I try to run a project that links to the edgetpu library I still get the `unresolved custom op` error. I've also tried running Namburger's [minimal example](https://github.com/Namburger/edgetpu-minimal-example), that I've gotten to work on ARMv7, on this Ubuntu 21.04 and it also got the same error. Any suggestions / advice?

manoj7410, 2021-06-16:
@BernardinD Are there any steps that you can share to reproduce this issue ? 
What is the python3version that you are working with?
Can you also share the detailed error including the command that you executed ?

dmitriykovalev, 2021-07-31:
Please add more details as @manoj7410 suggested, closing this for now.

closed 2021-07-31 by dmitriykovalev