google-coral/libedgetpu #22: Broken symbolic links after build completion

Doom4535, 2021-06-06:
When a run a build with Docker, it seems that final clean phase is incomplete, as several broken symbolic links are left behind:
```bash
ls -l
-------------------------------
lrwxrwxrwx  1 aaron aaron   111 Jun  6 03:22 bazel-bin -> /home/aaron/.cache/bazel/_bazel_aaron/eab0d61a99b6696edb3d2aff87b585e8/execroot/libedgetpu/bazel-out/k8-opt/bin
lrwxrwxrwx  1 aaron aaron   100 Jun  6 03:22 bazel-out -> /home/aaron/.cache/bazel/_bazel_aaron/eab0d61a99b6696edb3d2aff87b585e8/execroot/libedgetpu/bazel-out
lrwxrwxrwx  1 aaron aaron   116 Jun  6 03:22 bazel-testlogs -> /home/aaron/.cache/bazel/_bazel_aaron/eab0d61a99b6696edb3d2aff87b585e8/execroot/libedgetpu/bazel-out/k8-opt/testlogs
lrwxrwxrwx  1 aaron aaron    90 Jun  6 03:22 bazel-workspace -> /home/aaron/.cache/bazel/_bazel_aaron/eab0d61a99b6696edb3d2aff87b585e8/execroot/libedgetpu
```
```bash
find . -xtype l
-------------------------------
./bazel-out
./bazel-bin
./bazel-workspace
./bazel-testlogs
```

I suspect this is because bazel cleans the catch (in this case `eab0d61a99b6696edb3d2aff87b585e8`) after the build finishes successfully, which breaks the symbolic links.

Build command: `DOCKER_CPUS="k8" DOCKER_IMAGE="ubuntu:18.04" DOCKER_TARGETS=libedgetpu make docker-build`
Docker Version: `Docker version 20.10.6, build 370c28948e`

manoj7410, 2021-06-16:
@Doom4535 Are you able to navigate into these directories ?

mbrooksx, 2021-06-16:
Hello @Doom4535 ,

This working as intended - the symlinks are unfortunate though. The idea of building within the Docker container is to keep the Bazel workspace/outputs within that container. When running via docker-build, once the build completes the outputs are copied to the out/ folder that is external to the Docker container - the Bazel intermediate files are not.

If you want to see all these files you can jump into the docker container using:
```
make docker-shell
(inside Docker) make libedgetpu
```

closed 2021-06-25 by manoj7410