google-coral/libedgetpu #24: Disconnect every 5mn Virtualbox Win10 Host

Pascal66, 2021-07-08:
Dont know if it's related to the driver win10, virtual box usb driver, or kernel of the VM

Running on USB 3.0 (The PICe doesnt work with virtualbox)
Continuous error, starting from '`unable to get BOX...`'
[image]

Tryied slow or fast version, usb 2.0 or 3.0 port
(Dont know if it work on usb C port)

Pascal66, 2021-07-08:
I've stabilised things (65ms inference), using the maximum powermanagement of everything in win10 host.
Just the "`unable to get BOS descriptor or descriptor too short`" at initialisation, since  20mn
No more Transfert event error for now.

manoj7410, 2021-07-09:
@Pascal66 Are you able to run the device successfully on native/host OS installation instead of the VM ?

Pascal66, 2021-07-09:
@manoj7410 I've no dev environment in the W10 host.

Pascal66, 2021-07-13:
Some news, after tryied all energy saving combinations on w10 host, I put all to normal.
I do that because I suspect something else than energy saving causing my problems.
No more success, but that confirm that power saving on w10 host isnt the problem.

Then I changed my input video flux to a low resolution, and near all is working now.
(No more `comp_code`, no more fatal disconnecting usb)

Just the `BOS descriptor` exist when starting the CoralUSB, and a mean 40ms inference time

Pascal66, 2021-07-16:
More detail (in linux guest VM) : 
```
[25175.796563] xhci_hcd 0000:00:0c.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 2 comp_code 13
[25175.797153] xhci_hcd 0000:00:0c.0: Looking for event-dma 0000000001081e90 trb-start 0000000001081ce0 trb-end 0000000001081e20 seg-start 0000000001081000 seg-end 0000000001081ff0
[25206.590397] usb 2-1: reset SuperSpeed Gen 1 USB device number 20 using xhci_hcd
[25206.605977] usb 2-1: unable to get BOS descriptor or descriptor too short
[25206.617319] usb 2-1: device firmware changed
[25206.625480] usb 2-1: USB disconnect, device number 20
[25206.744047] usb 2-1: new SuperSpeed Gen 1 USB device number 21 using xhci_hcd
[25206.758982] usb 2-1: unable to get BOS descriptor or descriptor too short
[25206.773754] usb 2-1: New USB device found, idVendor=18d1, idProduct=9302, bcdDevice= 1.00
[25206.773760] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[25207.006742] usb 2-1: USB disconnect, device number 21
[25208.945130] usb 2-1: new SuperSpeed Gen 1 USB device number 22 using xhci_hcd
[25208.959934] usb 2-1: LPM exit latency is zeroed, disabling LPM.
[25208.961285] usb 2-1: New USB device found, idVendor=18d1, idProduct=9302, bcdDevice= 1.00
[25208.961290] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[25482.132130] xhci_hcd 0000:00:0c.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 2 comp_code 13
[25482.145226] xhci_hcd 0000:00:0c.0: Looking for event-dma 0000000001081b10 trb-start 0000000001080ac0 trb-end 0000000001080ac0 seg-start 0000000001080000 seg-end 0000000001080ff0
[25526.928499] usb 2-1: reset SuperSpeed Gen 1 USB device number 22 using xhci_hcd
[25526.942364] usb 2-1: unable to get BOS descriptor or descriptor too short
[25526.956185] usb 2-1: device firmware changed
[25526.965025] usb 2-1: USB disconnect, device number 22
[25527.081372] usb 2-1: new SuperSpeed Gen 1 USB device number 23 using xhci_hcd
[25527.117063] usb 2-1: unable to get BOS descriptor or descriptor too short
[25527.129497] usb 2-1: New USB device found, idVendor=18d1, idProduct=9302, bcdDevice= 1.00
[25527.129503] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[25527.338863] usb 2-1: USB disconnect, device number 23
[25528.581201] usb 2-1: new SuperSpeed Gen 1 USB device number 24 using xhci_hcd
[25528.599823] usb 2-1: LPM exit latency is zeroed, disabling LPM.
[25528.600483] usb 2-1: New USB device found, idVendor=18d1, idProduct=9302, bcdDevice= 1.00
[25528.600488] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[25817.550270] xhci_hcd 0000:00:0c.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 2 comp_code 1
[25817.562320] xhci_hcd 0000:00:0c.0: Looking for event-dma 0000000001081d10 trb-start 0000000001080cd0 trb-end 0000000001080cd0 seg-start 0000000001080000 seg-end 0000000001080ff0
```

Pascal66, 2021-07-27:
New version : 
no more dma errors for now.
Still Just the "unable to get BOS descriptor or descriptor too short" at initialisation