google-coral/libedgetpu #25: No module named 'pycoral.adapters'

arun-kumark, 2021-07-13:
Hello,
I am testing the M.2 Card Google Coral on Windows10.
Python version  3.8

While  executing an example of Parrot, I am facing the following issue:

```
python examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
Traceback (most recent call last):
  File "examples/classify_image.py", line 36, in <module>
    from pycoral.adapters import classify
ModuleNotFoundError: No module named 'pycoral.adapters'

B'''   But the module is already present .

I am not sure , what is wrong with tmy setup, could you help?

thanks 
Kind Regards
Arun

hjonnala, 2021-07-13:
please try uninstall tflite runtime and install from whl package from this source: https://github.com/google-coral/pycoral/releases

`pip install https://github.com/google-coral/pycoral/releases/download/v1.0.1/tflite_runtime-2.5.0-cp38-cp38-win_amd64.whl`

arun-kumark, 2021-07-13:
Thanks for the quick reply,
I did that, but still the same output.

Below is my pip freeze output. Do you see problem? 
```
google@DESKTOP-QNUDFH0 /cygdrive/c/Users/google/coral
$ pip freeze
absl-py==0.13.0
area==1.1.1
astunparse==1.6.3
beautifulsoup4==4.9.3
cachetools==4.2.2
certifi==2021.5.30
chardet==4.0.0
flatbuffers==1.12
gast==0.4.0
geojson==2.5.0
google-auth==1.32.1
google-auth-oauthlib==0.4.4
google-pasta==0.2.0
grpcio==1.34.1
h5py==3.1.0
idna==2.10
keras-nightly==2.5.0.dev2021032900
Keras-Preprocessing==1.1.2
Markdown==3.3.4
numpy==1.19.5
oauthlib==3.1.1
opt-einsum==3.3.0
Pillow==8.3.1
progressbar2==3.53.1
protobuf==3.17.3
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycoral==0.0.7
python-utils==2.5.6
requests==2.25.1
requests-oauthlib==1.3.0
rsa==4.7.2
six==1.15.0
soupsieve==2.2.1
tensorboard==2.5.0
tensorboard-data-server==0.6.1
tensorboard-plugin-wit==1.8.0
tensorflow==2.5.0
tensorflow-estimator==2.5.0
termcolor==1.1.0
tflite-runtime @ file:///C:/Users/google/coral/tflite_runtime-2.5.0-cp38-cp38-win_amd64.whl
typing-extensions==3.7.4.3
urllib3==1.26.6
Werkzeug==2.0.1
wrapt==1.12.1
```
Thanks

hjonnala, 2021-07-13:
ok..uninstall pycoarl also install from this it needs to 1.0.1 version.

`pip install https://github.com/google-coral/pycoral/releases/download/v1.0.1/pycoral-1.0.1-cp38-cp38-win_amd64.whl`

arun-kumark, 2021-07-13:
Thanks a lot, it worked.
I am not sure how it worked, but I did the installation just like you mentioned, I reinstalled (after uninstalling) and it worked. A small explanation will really helpful for future.

google@DESKTOP-QNUDFH0 /cygdrive/c/Users/google/coral
$ python examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
----INFERENCE TIME----
Note: The first inference on Edge TPU is slow because it includes loading the model into Edge TPU memory.
11.6ms
2.5ms
2.4ms
2.5ms
2.5ms
-------RESULTS--------
Ara macao (Scarlet Macaw): 0.75781

google@DESKTOP-QNUDFH0 /cygdrive/c/Users/google/coral
$
Thanks
Kind Regards
Arun

hjonnala, 2021-07-13:
The error came because the script not able to find adapter module inside site-packages/pycoral.

Now, you can check this path **/site-packages/pycoral**  you will find those adapter modules along with others.

closed 2021-07-13 by hjonnala

scottamain, 2021-07-29:
Hi @arun-kumark I'd like to understand how you got into this situation. Can you please tell me the exact command you used the first time you installed pycoral?

(You ended up installing this [other package called pycoral](https://pypi.org/project/pycoral/) and I want to figure out how that happened so we can help others avoid it.)

arun-kumark, 2021-08-09:
I replicated on the system freshly and ended up at the same error.
I am following : https://coral.ai/docs/m2/get-started/#4-run-a-model-on-the-edge-tpu
Please see below my steps on Debian 11.

```
google@google:~$ cd coral/
google@google:~/coral$ bash examples/install_requirements.sh classify_image.py
bash: examples/install_requirements.sh: No such file or directory
google@google:~/coral$ ls
pycoral
google@google:~/coral$ cd pycoral/
google@google:~/coral/pycoral$ bash examples/install_requirements.sh classify_image.py
DOWNLOAD: mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   181  100   181    0     0   2350      0 --:--:-- --:--:-- --:--:--  2381
100 4197k  100 4197k    0     0  3013k      0  0:00:01  0:00:01 --:--:-- 3538k
DOWNLOAD: mobilenet_v2_1.0_224_inat_bird_quant.tflite
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   173  100   173    0     0   2217      0 --:--:-- --:--:-- --:--:--  2217
100 3495k  100 3495k    0     0  1368k      0  0:00:02  0:00:02 --:--:-- 1595k
DOWNLOAD: inat_bird_labels.txt
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   150  100   150    0     0   2000      0 --:--:-- --:--:-- --:--:--  2000
100 37146  100 37146    0     0   209k      0 --:--:-- --:--:-- --:--:--  209k
DOWNLOAD: parrot.jpg
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   140  100   140    0     0   1944      0 --:--:-- --:--:-- --:--:--  1944
100  582k  100  582k    0     0  1735k      0 --:--:-- --:--:-- --:--:-- 4941k
google@google:~/coral/pycoral$ python3 examples/classify_image.py \
--model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite \
--labels test_data/inat_bird_labels.txt \
--input test_data/parrot.jpg
Traceback (most recent call last):
  File "examples/classify_image.py", line 37, in <module>
    from pycoral.adapters import classify
ModuleNotFoundError: No module named 'pycoral'
google@google:~/coral/pycoral$ cd ..
```
Thanks
Kind Regards
Arun

scottamain, 2021-08-09:
This looks like a different error because it says you don't have `pycoral` at all, whereas the original error you had was because Python could not find `pycoral.adapters`... Your original problem was you installed a different Python package that's *also called pycoral* and it does not have the `adapters` submodule. But now, you don't have *any* module named `pycoral`.

You can be sure to get our package by specifying the pycoral version like this:

```
python3 -m pip install --extra-index-url https://google-coral.github.io/py-repo/ pycoral~=2.0
```

But I'd still like to understand your system environment to diagnose how you got the wrong package the first time... Please run this command in a terminal and tell me the output:

```
python3 -c "import distutils.util;print(distutils.util.get_platform().replace('-', '_'))"
```