google-coral/libedgetpu #27: edgetpu on Fedora 34 Bazel 3.7.1

pmdaye, 2021-07-26:
Hello,

I am trying to compile the edge TPU library for my fedora 34.
Using the installation process of bazel for fedora ([webpage](https://docs.bazel.build/versions/4.1.0/install-redhat.html)), I install bazel 3.7.1
I used the master of libedgetpu and as requested, I modified the value for the commit and the SHA256 (corresponding to tf 2.5.0) from 
 
> ! TENSORFLOW_COMMIT = "a4dfb8d1a71385bd6d122e4f27f86dcebb96712d"
> ! TENSORFLOW_SHA256 = "cb99f136dc5c89143669888a44bfdd134c086e1e2d9e36278c1eb0f03fe62d76"

to

> ! TENSORFLOW_COMMIT = "855c4c0ee34257b98ce2d01121940efb5423a059"
> ! TENSORFLOW_SHA256 = "855ded5cae5915c6f232d27342fa4fb666922798e02e15379b4fa67265695685"

Then, I tried to run make, the downloads worked nicely but I have an error message when I build:

>ERROR: /home/pdaye/.cache/bazel/_bazel_pdaye/76fec2d0d26d314733c7d17e3ea4f44e/external/flatbuffers/src/BUILD:40:11: >undeclared inclusion(s) in rule '@flatbuffers//src:flatc':
>this rule is missing dependency declarations for the following files included by 'flatbuffers/src/util.cpp':
>  '/usr/lib/gcc/x86_64-redhat-linux/11/include/limits.h'
>  '/usr/lib/gcc/x86_64-redhat-linux/11/include/syslimits.h'
>  '/usr/lib/gcc/x86_64-redhat-linux/11/include/stdint.h'
>  '/usr/lib/gcc/x86_64-redhat-linux/11/include/stddef.h'
>  '/usr/lib/gcc/x86_64-redhat-linux/11/include/stdarg.h'

Any chance someone has an idea on why this happens and how I could fix it?
Thank you very much!

manoj7410, 2021-07-28:
@pmdaye Did you try to build with Docker ? Try to build using the Docker as recommended here : https://github.com/google-coral/libedgetpu#building

pmdaye, 2021-07-28:
@manoj7410 Thanks for the reply. This was not working. However, the approach with the Makefile only worked smoothly. I am now in the process of writing a CMakeLists.txt to allow compiling and installing with CMake such that I can include it more easily in my build (within a Yocto recipe).
Thanks,

dmitriykovalev, 2021-07-29:
@pmdaye Can you please try to build again with Bazel using the latest code from `master`? I believe https://github.com/google-coral/libedgetpu/commit/0f1def1f0e0e5ee79a517f30733f09dda2441304 fixes the problem.

pmdaye, 2021-07-30:
@dmitriykovalev I will try later as I have a CMakeLists.txt that works nicely and can be integrated with my embedded distribution (Yocto).

manoj7410, 2021-08-03:
Feel free to reopen this thread if the issue still persists.

closed 2021-08-03 by manoj7410