google-coral/libedgetpu #29: edgetpu_runtime_20210726 not working (Win10)

agjunyent, 2021-08-31:
I have updated the runtime to the latest version on my Windows 10 machine, and now the coral does not even start.

I have a simple code I was using to test the coral models (because some times the compiler simply does not work):

```
...
current_model = tflite.Interpreter(model_path="model.tflite",
                                                   experimental_delegates=[tflite.load_delegate('edgetpu.dll')])
current_model.allocate_tensors()
...
```

I cannot make it go pass the "tflite.Interpreter(" call. It simply goes there, I can hear the sound of the USB device "disconnecting", but it never connects, and pyhton just ends there.

Tried to go back to old version but new models are now compiled against the new one, so I cannot make it run. Any help?

hjonnala, 2021-08-31:
Hi @agjunyent yes, edgetpu runtime 14 is not working on windows. Due to the other priorities this issue won't be fixed in the near future. Please use edgetpu_runtime_20210119.zip from https://coral.ai/software/#edgetpu-runtime and compile the models with runtime version 13 by specifying -m flag.

To know more about the details of flags please refer to this resource: https://coral.ai/docs/edgetpu/compiler/#usage

`! edgetpu_compiler -s -m 13 ssdlite_mobiledet_dog_vs_cat.tflite`

agjunyent, 2021-08-31:
thanks for your answer @hjonnala !!

I'm moving the test enviromnent to a RPi (the production environment is indeed a RPi, was using windows as first test) to test it against it. Is it working on the RPi?

thanks

hjonnala, 2021-08-31:
Yes, Edgetpu runtime 14 is working on RPi.

closed 2021-08-31 by agjunyent