google-coral/libedgetpu #33: some weird malloc error when calling GetSingleton()->OpenDevice()

anuragkarwa001, 2021-11-23:
### Description

I have created a sample program and was trying detection with a single edge tpu and it was working fine. But when I added some code that basically parses the model name and labels file from  a config file the function call gives a malloc error. But as far as I can see I have not passes anything to this function as argument and also I have checked that the model is present on the location provided .

edgeTpuContext = edgetpu::EdgeTpuManager::GetSingleton()->OpenDevice();
the function call above give the error 
malloc.c:4036: _int_malloc: Assertion `(unsigned long) (size) >= (unsigned long) (nb)' failed.

### Issue Type

Bug

### Operating System

Ubuntu

### Coral Device

Mini PCIe

### Programming Language

C++

### Relevant Log Output

```shell
malloc.c:4036: _int_malloc: Assertion `(unsigned long) (size) >= (unsigned long) (nb)' failed.
```

hjonnala, 2021-11-23:
@anuragkarwa001 can you try to pass the parameters as input arguments to the script.

hjonnala, 2021-12-09:
closing due to inactivity. Feel free to reopen if you are still having the issue. Thanks!

closed 2021-12-13 by hjonnala