google-coral/libedgetpu #35: Interpreter->invoke() calls Segmentation Fault

enfild, 2021-12-09:
### Description

**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Linux Ubuntu 18.04 with USB Coral. 
- TensorFlow installed from (source or binary): source
- TensorFlow version (or github SHA if from source): v2.5.0, v2.6.0

**Description**
When i calls "initTfLiteInterpreter", tfLite work is correct, output has all information (boxes, labels and classes), time of invoke is good (15 ms). But when i calls "processingFrame" in other classes with equalent code (i get Seg fault on "Interpreter->invoke()"):
```
for(int i = 0; i< 10; i++)
    {
        cv::Mat testImage = cv::imread(EXAMPLE_FRAME);
        processingFrame(testImage);
    }
```
I get this error with TF 2.5.0, 2.6.0. 

**Source of my programm**
**.h:**
```
/// Build EDGE Interpreter for Coral
    void BuildEdgeTpuInterpreter(const tflite::FlatBufferModel &model,
                                edgetpu::EdgeTpuContext *edgetpu_context);

    // Load graph to coral
    void initTfLiteInterpreter();

    // Processing the received frame
    void processingFrame(cv::Mat& frame);

    int num_threads = 1;
    std::unique_ptr<tflite::Interpreter> interpreter;
    std::shared_ptr<edgetpu::EdgeTpuContext> tpu_context;

    TfLiteTensor* input_tensor;
    TfLiteTensor* output_locations;
    TfLiteTensor* output_classes;
    TfLiteTensor* output_scores;
    TfLiteTensor* num_detections_;

    int height;
    int width;
    int channels;
    int row_elems;
```
**.cxx:**
```
void ControllerNeuroNetworksPrivate::BuildEdgeTpuInterpreter(const tflite::FlatBufferModel &model,
                                                               edgetpu::EdgeTpuContext *edgetpu_context)
{
    tflite::ops::builtin::BuiltinOpResolver resolver;
    resolver.AddCustom(edgetpu::kCustomOp, edgetpu::RegisterCustomOp());
    if (tflite::InterpreterBuilder(model, resolver)(&interpreter) != kTfLiteOk) {
        std::cerr << "Failed to build interpreter." << std::endl;
        return;
    }
    // Allocate tensor buffers.
    // Bind given context with interpreter.
    interpreter->SetExternalContext(kTfLiteEdgeTpuContext, edgetpu_context);
    interpreter->SetNumThreads(1);
    if (interpreter->AllocateTensors() != kTfLiteOk)
    {
      std::cerr << "Failed to allocate tensors." << std::endl;
    }
}

void ControllerNeuroNetworksPrivate::initTfLiteInterpreter(void)
{
    auto model = tflite::FlatBufferModel::BuildFromFile(GRAPH.c_str());

    tpu_context = edgetpu::EdgeTpuManager::GetSingleton()->OpenDevice();
    std::cout << "Checking readiness of Coral device" << std::endl;
    if(!tpu_context->IsReady())
    {
        std::cout << "Coral device is not ready" << std::endl;
        throw -1;
    }
    std::cout << "EDGE TPU path: " << tpu_context->GetDeviceEnumRecord().path << std::endl;
    BuildEdgeTpuInterpreter(*model, tpu_context.get());

    input_tensor = interpreter->tensor(interpreter->inputs()[0]);
    output_locations = interpreter->tensor(interpreter->outputs()[0]);
    output_classes = interpreter->tensor(interpreter->outputs()[1]);
    output_scores = interpreter->tensor(interpreter->outputs()[2]);
    num_detections_ = interpreter->tensor(interpreter->outputs()[3]);

    height = input_tensor->dims->data[1];
    width = input_tensor->dims->data[2];
    channels = input_tensor->dims->data[3];
    row_elems = width * channels;

    for(int i = 0; i< 10; i++)
    {
        cv::Mat testImage = cv::imread(EXAMPLE_FRAME);
        processingFrame(testImage);
    }
    
    Utils::dual_write("CNN is ready, example frame was processed");
    m_readyFlag.store(true);
}

void ControllerNeuroNetworksPrivate::processingFrame(cv::Mat& frame)
{
    Q_ASSERT(q_ptr);
    const clock_t begin_time = clock();
    QMutexLocker locker(&m_mutex);
    qDebug() << "cv mat size: " << width << height;
    cvtColor(frame, frame, cv::COLOR_BGR2RGB);
    // Resize for model input
    cv::resize(frame, frame, cv::Size(width, height));

    if (input_tensor->type != kTfLiteUInt8 ||           //
        input_tensor->dims->data[0] != 1 ||             //
        input_tensor->dims->data[1] != height ||  //
        input_tensor->dims->data[2] != width ||   //
        input_tensor->dims->data[3] != channels) {
    std::cerr << "Input tensor shape does not match input image" << std::endl;
    return;
    }

    uint8_t* dst = input_tensor->data.uint8;
    for (int row = 0; row < height; row++) {
        memcpy(dst, frame.ptr(row), row_elems);
        dst += row_elems;
    }

    if(interpreter->Invoke() != kTfLiteOk)
        qDebug() << "Invoke is broken";
    qDebug() << "Invoke is done!";
    const float* detection_locations = output_locations->data.f;
    const float* detection_classes = output_classes->data.f;
    const float* detection_scores = output_scores->data.f;
    const int num_detections = *(num_detections_->data.f);
    for (int i = 0; i < num_detections; i++) {
        const float score = detection_scores[i];
        const std::string label = std::to_string(uint8_t(detection_classes[i]));
        const float yMin = detection_locations[4 * i + 0];
        const float xMin = detection_locations[4 * i + 1];
        const float yMax = detection_locations[4 * i + 2];
        const float xMax = detection_locations[4 * i + 3];
        if (score > thresholdScore) {
            std::cout << label << " score:" << score << std::endl;
            emit q_ptr->returnBoundingBoxes(frame, yMin, xMin, yMax, xMax, score, label, true);
        }
    }
    std::cout << "time: " << float( clock () - begin_time ) /  CLOCKS_PER_SEC << std::endl;
    emit q_ptr->finishedCNNProcessing(frame);
}
```

### Issue Type

Bug

### Operating System

Ubuntu

### Coral Device

USB Accelerator

### Programming Language

C++

### Relevant Log Output

```shell
**LOGS**

Checking readiness of Coral device
EDGE TPU path: /sys/bus/usb/devices/2-1
cv mat size:  640 480
Invoke is done!
1 score:0.902344
time: 0.022215
cv mat size:  640 480
Invoke is done!
1 score:0.902344
time: 0.012465
cv mat size:  640 480
Invoke is done!
1 score:0.902344
time: 0.011841
cv mat size:  640 480
Invoke is done!
1 score:0.902344
time: 0.011659
cv mat size:  640 480
Invoke is done!
1 score:0.902344
time: 0.014413
cv mat size:  640 480
Invoke is done!
1 score:0.902344
time: 0.011502
cv mat size:  640 480
Invoke is done!
1 score:0.902344
time: 0.012496
cv mat size:  640 480
Invoke is done!
1 score:0.902344
time: 0.012136
cv mat size:  640 480
Invoke is done!
1 score:0.902344
time: 0.012898
cv mat size:  640 480
Invoke is done!
1 score:0.902344
time: 0.012129
Thu Dec  9 11:52:59 2021:  CNN is ready, example frame was processed
cv mat size:  640 480
Segmentation fault (core dumped)
```
```

hjonnala, 2021-12-09:
Hi @enfild are using [edgetpu_runtime 14](https://coral.ai/software/#edgetpu-runtime) version?

Can you try with any other models and let us know if it's happening with other models too. Thanks!

hjonnala, 2021-12-09:
I think, it is okay if you build the library using [libedgetpu](https://github.com/google-coral/libedgetpu) repo. Just want to make sure, if you are using latest runtime.

Here are some object detection models: https://coral.ai/models/object-detection/
You can get all those models from [test_data](https://github.com/google-coral/test_data) repo.

enfild, 2021-12-10:
@hjonnala Thank you for help, this issue based on uncorrect model, because models from test_repo by coral, work is good. i close issue

enfild, 2021-12-13:
Hi @hjonnala . Unfortunately I was happy early, because I checked the models from the test data only for classification problems. But when I take models to object detection, the problem comes back again. May by you has any idea?

hjonnala, 2021-12-13:
Hi @emfid can you please share the files and steps needed to reproduce the issue.

enfild, 2021-12-13:
@hjonnala it is module for main programm
Files:
https://drive.google.com/drive/folders/1WPE7orTM55xZe7Hkz5r4z7xYrZ5oHSF3?usp=sharing

hjonnala, 2021-12-13:
Thanks for sharing the files. I am trying to build the code with cmake but i am getting the following error. Can you lease share the steps how build and run the code. Also,  please share the model that you are trying to run. Thanks!

<details>

```
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenCV: /usr (found version "4.5.3") 
-- OpenCV_INCLUDE_DIRS = /usr/include/opencv4
-- OpenCV_LIBS = opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_stitching;opencv_video;opencv_videoio;opencv_alphamat;opencv_aruco;opencv_barcode;opencv_bgsegm;opencv_bioinspired;opencv_ccalib;opencv_datasets;opencv_dnn_objdetect;opencv_dnn_superres;opencv_dpm;opencv_face;opencv_freetype;opencv_fuzzy;opencv_hdf;opencv_hfs;opencv_img_hash;opencv_intensity_transform;opencv_line_descriptor;opencv_mcc;opencv_optflow;opencv_phase_unwrapping;opencv_plot;opencv_quality;opencv_rapid;opencv_reg;opencv_rgbd;opencv_saliency;opencv_shape;opencv_stereo;opencv_structured_light;opencv_superres;opencv_surface_matching;opencv_text;opencv_tracking;opencv_videostab;opencv_viz;opencv_wechat_qrcode;opencv_ximgproc;opencv_xobjdetect;opencv_xphoto
CMake Error at CMakeLists.txt:51 (qt5_use_modules):
  Unknown CMake command "qt5_use_modules".

```
</details>

enfild, 2021-12-14:
@hjonnala I wrote the required minimal example, to use it you need to run the build from the test folder(with model and image). 
https://drive.google.com/drive/folders/1jvRUpYHemYuGA56wWdccy4Jsc8v-0biv?usp=sharing
Please install Qt:
```
sudo apt install qt5-default
```

hjonnala, 2021-12-14:
I am having issues with qt installation on linux. But I am able to run benchmarks using [libcoral](https://github.com/google-coral/libcoral/blob/master/coral/single_model_benchmark.cc) and object detection using [PyCoral API](https://github.com/google-coral/pycoral/blob/master/examples/detect_image.py). Looks like the model is not detecting the objects accurately. Have you tested the accuracy with CPU tflite model? can you you please try your scripts with this [model](https://github.com/google-coral/test_data/blob/master/ssd_mobilenet_v2_coco_quant_postprocess_edgetpu.tflite) and let me know if you are facing any issues?

```
hemanth@hemanth-glaptop:~/coral/pycoral$ python3 examples/detect_image.py   --model ../../Downloads/NNcontroller-20211214T134511Z-001/NNcontroller/test/detect_edgetpu.tflite   --labels test_data/coco_labels.txt   --input test_data/grace_hopper.bmp   --output ${HOME}/grace_hopper_processed.bmp -c 5
----INFERENCE TIME----
Note: The first inference is slow because it includes loading the model into Edge TPU memory.
25.41 ms
14.82 ms
14.64 ms
14.83 ms
15.19 ms
-------RESULTS--------
bicycle
  id:     1
  score:  0.5703125
  bbox:   BBox(xmin=-27, ymin=87, xmax=468, ymax=623)
person
  id:     0
  score:  0.4296875
  bbox:   BBox(xmin=316, ymin=478, xmax=463, ymax=540)
```
```
hemanthj@hemanth-glaptop:~/libcoral/out/k8/benchmarks/coral$ ./single_model_benchmark -model ../../../../..//Downloads/NNcontroller-20211214T134511Z-001/NNcontroller/test/detect_edgetpu.tflite
2021-12-14 14:14:17
Running ./single_model_benchmark
Run on (4 X 3900 MHz CPU s)
CPU Caches:
  L1 Data 32K (x2)
  L1 Instruction 32K (x2)
  L2 Unified 256K (x2)
  L3 Unified 4096K (x1)
Load Average: 3.91, 4.35, 3.56
***WARNING*** CPU scaling is enabled, the benchmark real time measurements may be noisy and will incur extra overhead.
-----------------------------------------------------
Benchmark           Time             CPU   Iterations
-----------------------------------------------------
BM_Model         14.8 ms         2.31 ms          290 detect_edgetpu.tflite
```

enfild, 2021-12-15:
@hjonnala I think we have some kind of misunderstanding. Thanks for linking to the model, i tested and gets the same error.
My model may be inaccurate, the quality is poor, but it doesn't matter to me. It is important for me to understand why I get a segmentation error when embedding Coral with TF in projects.
i removed Qt from proj:
https://drive.google.com/drive/folders/1sdBT6NsJnocuPagVFxDLuaPm7cGcT2yk?usp=sharing

enfild, 2021-12-17:
Hi @hjonnala . I'm sorry, I just noticed that the access was not open. Corrected.

enfild, 2021-12-20:
Valgrind output is `Conditional jump or move depends on uninitialised value(s)`
 ```
valgrind ./SkyDanceServer==29672== Memcheck, a memory error detector
==29672== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==29672== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==29672== Command: ./SkyDanceServer
==29672== 
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
--29672-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x13
Loading config: config.json
FrameReceiverController
FrameReceiverControllerPrivate
USB or CSI camera
[ WARN:0] global /home/pavel/Downloads/opencv-4.5.1/modules/videoio/src/cap_gstreamer.cpp (961) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
start ControllerNeuroNetworks
start ControllerNeuroNetworksPrivate
Checking readiness of Coral device
EDGE TPU path: /sys/bus/usb/devices/2-1
cv mat size:  640 480
Coral is ready!
Invoke is done!
time: 55.3672
cv mat size:  640 480
Coral is ready!
Invoke is done!
time: 159.751
Mon Dec 20 15:08:33 2021:  CNN is ready, example frame was processed
SaveFrameClass
cv mat size:  640 480
Coral is ready!
==29672== Thread 7 FrameReceiverCon:
==29672== Conditional jump or move depends on uninitialised value(s)
==29672==    at 0x67E66C: tflite::ops::custom::detection_postprocess::ValidateBoxes(TfLiteTensor const*, int) (in /home/pavel/visinav-rnd/build/SkyDanceServer/SkyDanceServer)
==29672==    by 0x68006E: tflite::ops::custom::detection_postprocess::NonMaxSuppressionSingleClassHelper(TfLiteContext*, TfLiteNode*, tflite::ops::custom::detection_postprocess::OpData*, std::vector<float, std::allocator<float> > const&, std::vector<int, std::allocator<int> >*, int) (in /home/pavel/visinav-rnd/build/SkyDanceServer/SkyDanceServer)
==29672==    by 0x681143: tflite::ops::custom::detection_postprocess::NonMaxSuppressionMultiClassFastHelper(TfLiteContext*, TfLiteNode*, tflite::ops::custom::detection_postprocess::OpData*, float const*) (in /home/pavel/visinav-rnd/build/SkyDanceServer/SkyDanceServer)
==29672==    by 0x68169A: tflite::ops::custom::detection_postprocess::NonMaxSuppressionMultiClass(TfLiteContext*, TfLiteNode*, tflite::ops::custom::detection_postprocess::OpData*) (in /home/pavel/visinav-rnd/build/SkyDanceServer/SkyDanceServer)
==29672==    by 0x56F056: tflite::Subgraph::Invoke() (in /home/pavel/visinav-rnd/build/SkyDanceServer/SkyDanceServer)
==29672==    by 0x49E092: tflite::Interpreter::Invoke() (in /home/pavel/visinav-rnd/build/SkyDanceServer/SkyDanceServer)
```

manoj7410, 2021-12-20:
@enfild Did you try to build the libedgetpu with debug/detailed logging enabled? We can get some hints from there. I would suggest you to change the log_level of libedgetpu at https://github.com/google-coral/libedgetpu/blob/master/port/default/port_from_tf/logging.cc#L171 and then rebuild this library by following the instructions given at https://github.com/google-coral/libedgetpu#building.

You can set the log_level value from 1 to 10 (10 being the most detailed) to get the detailed output of libedgetpu and then we can probably understand where exactly it is failing.

enfild, 2021-12-20:
@manoj7410 thanks for you recomendetions!
Now i get this output:
<details>I driver/usb/local_usb_device.cc:672] ASYNC IN 1 end
I driver/usb/usb_driver.cc:1138] bulk in 1024 bytes from buffer index [0]
I driver/usb/usb_driver.cc:1138] bulk in 1024 bytes from buffer index [1]
I driver/usb/usb_driver.cc:1138] bulk in 1024 bytes from buffer index [2]
I driver/usb/usb_driver.cc:1138] bulk in 1024 bytes from buffer index [3]
I driver/usb/usb_driver.cc:1138] bulk in 1024 bytes from buffer index [4]
I driver/usb/usb_driver.cc:1138] bulk in 1024 bytes from buffer index [5]
I driver/dma_chunker.cc:30] Completed 0 bytes; Outstanding 0 bytes; Processing next 48384 bytes
I driver/usb/local_usb_device.cc:642] UnregisterCompletedTransfer
I driver/usb/usb_driver.cc:916] [4-3] bulk in for 48384 bytes has yielded 1024 bytes from index [0]
I driver/dma_chunker.cc:30] Completed 1024 bytes; Outstanding 0 bytes; Processing next 47360 bytes
I driver/usb/usb_driver.cc:916] [4-3] bulk in for 47360 bytes has yielded 1024 bytes from index [1]
I driver/dma_chunker.cc:30] Completed 2048 bytes; Outstanding 0 bytes; Processing next 46336 bytes
I driver/usb/usb_driver.cc:916] [4-3] bulk in for 46336 bytes has yielded 1024 bytes from index [2]
I driver/dma_chunker.cc:30] Completed 3072 bytes; Outstanding 0 bytes; Processing next 45312 bytes
I driver/usb/local_usb_device.cc:672] ASYNC IN 1 end
I driver/usb/local_usb_device.cc:642] UnregisterCompletedTransfer
I driver/usb/usb_driver.cc:916] [4-3] bulk in for 45312 bytes has yielded 1024 bytes from index [3]
I driver/usb/local_usb_device.cc:672] ASYNC IN 1 end
I driver/usb/local_usb_device.cc:642] UnregisterCompletedTransfer
I driver/dma_chunker.cc:30] Completed 4096 bytes; Outstanding 0 bytes; Processing next 44288 bytes
I driver/usb/usb_driver.cc:916] [4-3] bulk in for 44288 bytes has yielded 1024 bytes from index [4]
I driver/dma_chunker.cc:30] Completed 5120 bytes; Outstanding 0 bytes; Processing next 43264 bytes
I driver/usb/usb_driver.cc:916] [4-3] bulk in for 43264 bytes has yielded 1024 bytes from index [5]
I driver/usb/usb_driver.cc:1310] WorkerThreadFunc re-evaluation is needed
I driver/usb/usb_driver.cc:1174] WorkerThreadFunc dispatching 2 callback events in worker thread
I driver/usb/usb_driver.cc:1138] bulk in 1024 bytes from buffer index [6]
I driver/usb/usb_driver.cc:1138] bulk in 1024 bytes from buffer index [7]
I driver/usb/usb_driver.cc:1260] WorkerThreadFunc Installing bulk-in reader. buffer index [0]
I driver/usb/local_usb_device.cc:748] AsyncBulkInTransfer
I driver/usb/local_usb_device.cc:761] ASYNC IN 1 begin
I driver/usb/usb_driver.cc:1260] WorkerThreadFunc Installing bulk-in reader. buffer index [1]
I driver/usb/local_usb_device.cc:748] AsyncBulkInTransfer
I driver/usb/local_usb_device.cc:761] ASYNC IN 1 begin
I driver/usb/usb_driver.cc:1260] WorkerThreadFunc Installing bulk-in reader. buffer index [2]
I driver/usb/local_usb_device.cc:748] AsyncBulkInTransfer
I driver/usb/local_usb_device.cc:761] ASYNC IN 1 begin
I driver/usb/usb_driver.cc:1260] WorkerThreadFunc Installing bulk-in reader. buffer index [3]
I driver/usb/local_usb_device.cc:748] AsyncBulkInTransfer
I driver/usb/local_usb_device.cc:761] ASYNC IN 1 begin
I driver/usb/usb_driver.cc:1260] WorkerThreadFunc Installing bulk-in reader. buffer index [4]
I driver/usb/local_usb_device.cc:748] AsyncBulkInTransfer
I driver/usb/local_usb_device.cc:761] ASYNC IN 1 begin
I driver/usb/usb_driver.cc:1260] WorkerThreadFunc Installing bulk-in reader. buffer index [5]
I driver/usb/local_usb_device.cc:748] AsyncBulkInTransfer
I driver/usb/local_usb_device.cc:672] ASYNC IN 1 end
I driver/usb/local_usb_device.cc:642] UnregisterCompletedTransfer
I driver/usb/local_usb_device.cc:672] ASYNC IN 1 end
I driver/usb/local_usb_device.cc:642] UnregisterCompletedTransfer
I driver/usb/local_usb_device.cc:672] ASYNC IN 1 end
I driver/usb/local_usb_device.cc:642] UnregisterCompletedTransfer
I driver/usb/local_usb_device.cc:672] ASYNC IN 1 end
I driver/usb/local_usb_device.cc:642] UnregisterCompletedTransfer
I driver/usb/local_usb_device.cc:672] ASYNC IN 1 end
I driver/usb/local_usb_device.cc:642] UnregisterCompletedTransfer
I driver/usb/local_usb_device.cc:761] ASYNC IN 1 begin
I driver/dma_chunker.cc:30] Completed 6144 bytes; Outstanding 0 bytes; Processing next 42240 bytes
I driver/usb/local_usb_device.cc:672] ASYNC IN 1 end
I driver/usb/usb_driver.cc:916] [4-3] bulk in for 42240 bytes has yielded 1024 bytes from index [6]
I driver/dma_chunker.cc:30] Completed 7168 bytes; Outstanding 0 bytes; Processing next 41216 bytes
I driver/usb/usb_driver.cc:916] [4-3] bulk in for 41216 bytes has yielded 1024 bytes from index [7]
I driver/usb/usb_driver.cc:1310] WorkerThreadFunc re-evaluation is needed
I driver/usb/usb_driver.cc:1174] WorkerThreadFunc dispatching 6 callback events in worker thread
I driver/usb/usb_driver.cc:1138] bulk in 1024 bytes from buffer index [0]
I driver/usb/local_usb_device.cc:642] UnregisterCompletedTransfer
I driver/usb/usb_driver.cc:1138] bulk in 1024 bytes from buffer index [1]
I driver/usb/usb_driver.cc:1138] bulk in 1024 bytes from buffer index [2]
I driver/usb/usb_driver.cc:1138] bulk in 1024 bytes from buffer index [3]
[... 535 lines cut ...]
I driver/dma_chunker.cc:30] Completed 23552 bytes; Outstanding 0 bytes; Processing next 640 bytes
I driver/usb/usb_driver.cc:916] [5-3] bulk in for 640 bytes has yielded 640 bytes from index [7]
I driver/usb/usb_driver.cc:1310] WorkerThreadFunc re-evaluation is needed
[... cut ...]
I driver/single_queue_dma_scheduler.cc:234] Request[18]: Completed
I driver/single_tpu_request.cc:96] [18] Request destroyed.
I driver/usb/usb_driver.cc:550] IO completed
I driver/usb/usb_driver.cc:1310] WorkerThreadFunc re-evaluation is needed
I driver/usb/usb_driver.cc:1174] WorkerThreadFunc dispatching 0 callback events in worker thread
I driver/usb/usb_driver.cc:1317] WorkerThreadFunc waiting on state change
I driver/usb/usb_driver.cc:91] Unlocks both mutex
Segmentation fault (core dumped)
</details>

manoj7410, 2021-12-22:
I see that you have already used GDB. Did you try to run the bt command?
Also, can you try to run gdb ./<Executable> core

enfild, 2021-12-22:
@manoj7410  , yes i get `bt` out like this:
<detail>
I driver/usb/usb_driver.cc:1174] WorkerThreadFunc dispatching 0 callback events in worker thread
I driver/usb/usb_driver.cc:1317] WorkerThreadFunc waiting on state change
I driver/usb/usb_driver.cc:91] Unlocks both mutex

Thread 1 "ControllerNN_te" received signal SIGSEGV, Segmentation fault.
0x000000000063d89c in tflite::ops::custom::detection_postprocess::DecodeCenterSizeBoxes(TfLiteContext*, TfLiteNode*, tflite::ops::custom::detection_postprocess::OpData*) ()
(gdb) bt
#0  0x000000000063d89c in tflite::ops::custom::detection_postprocess::DecodeCenterSizeBoxes(TfLiteContext*, TfLiteNode*, tflite::ops::custom::detection_postprocess::OpData*) ()
#1  0x000000000063fa46 in tflite::ops::custom::detection_postprocess::Eval(TfLiteContext*, TfLiteNode*) ()
#2  0x0000000000537f81 in tflite::Subgraph::Invoke() ()
#3  0x000000000046826f in tflite::Interpreter::Invoke() ()
#4  0x0000000000465693 in ControllerNeuroNetworksPrivate::processingFrame (this=0xae40c0, frame=...)
    at /home/pavel/NNcontroller/src/ControllerNN_p.cxx:106
#5  0x0000000000464940 in ControllerNeuroNetworks::processingFrameOfCNN (this=0x7fffffffdd68, Frame=...)
    at /home/pavel/NNcontroller/src/ControllerNN.cxx:23
#6  0x000000000046237b in main (argc=1, argv=0x7fffffffdef8)
    at /home/pavel/NNcontroller/test/main.cxx:13
(gdb) 

</detail>

enfild, 2022-01-19:
I managed to solve the problem. using a shared pointer to the interpreter. And passing the pointer to the invoke execution area.

closed 2022-01-19 by enfild