google-coral/libedgetpu #36: Need a way to deallocate libusb delegates "leaked" after Python process is killed

talcs, 2021-12-15:
From experience on the Windows10 machine I work on, any USB socket becomes unusable for Coral USB stick if the Python/TFLite process that ran it has been killed - the OS resources stay occupied and cannot be allocated again for a given USB socket. 

The only workaround I found so far is once the USB socket cannot accept Coral anymore (getting an error from edgetpu.dll), is moving Coral to another USB socket which hasn't become unusable yet, or restarting the machine, which is many times very unwanted.

Those unordinary process kills can happen while debugging, finally forcing to restart the computer when all USB sockets are allocated forever... It will be great if a script is supplied to deallocate all "leaked" allocated delegates using libusb.

hjonnala, 2021-12-15:
HI @talcs  can you try the way 2 from this link: https://www.partitionwizard.com/partitionmanager/how-do-i-reset-my-usb-ports.html

talcs, 2021-12-16:
Hi @hjonnala , Thank you, but sadly it didn't help... The value in my registry was already 3. Every single USB port works until the first time the application crashes abruptly without deallocating the resource.... Need to somehow access those runaway living delegates and deallocate them.