google-coral/libedgetpu #40: error cross compilation for aarch64

liviolima80, 2022-05-03:
### Description

Hi,
I'm trying to cross-compile libedgetpu for aarch64 using the makefile approach in order to update the lib on coral dev board.
I used the suggested commit for TF2.5. I aldo downloaded the source code for abseil-cpp, flatbuffers and libusb that are the dependecies and edited the makefile_build/Makefile in order to add the include path of these libraries.

I have a bounch of error compiling $(TFROOT)/tensorflow/lite/util.cc : 

error: no matching function for call to 'tflite::Pool2DOptions::VerifyField<int8_t>(flatbuffers::Verifier&, tflite::Pool2DOptions::FlatBuffersVTableOffset)

error: no matching function for call to 'tflite::DepthwiseConv2DOptions::VerifyField<int8_t>(flatbuffers::Verifier&, tflite::DepthwiseConv2DOptions::FlatBuffersVTableOffset) const'

and so on..

Can someone help me?

Regards

### Issue Type

Support

### Operating System

Mendel Linux, Linux, Ubuntu

### Coral Device

Dev Board

### Programming Language

C++

### Relevant Log Output

```shell
Compiling /data/tensorflow/tensorflow/lite/util.cc
In file included from /data/tensorflow/tensorflow/lite/util.cc:30:
/data/tensorflow/tensorflow/lite/schema/schema_generated.h: In member function 'bool tflite::QuantizationParameters::Verify(flatbuffers::Verifier&) const':
/data/tensorflow/tensorflow/lite/schema/schema_generated.h:3241:58: error: no matching function for call to 'tflite::QuantizationParameters::VerifyField<uint8_t>(flatbuffers::Verifier&, tflite::QuantizationParameters::FlatBuffersVTableOffset) const'
            VerifyField<uint8_t>(verifier, VT_DETAILS_TYPE) &&
                                                          ^
In file included from /data/flatbuffers/include/flatbuffers/flatbuffer_builder.h:32,
                 from /data/flatbuffers/include/flatbuffers/flatbuffers.h:27,
                 from /data/tensorflow/tensorflow/lite/schema/schema_generated.h:21,
                 from /data/tensorflow/tensorflow/lite/util.cc:30:
/data/flatbuffers/include/flatbuffers/table.h:115:8: note: candidate: 'template<class T> bool flatbuffers::Table::VerifyField(const flatbuffers::Verifier&, flatbuffers::voffset_t, size_t) const'
   bool VerifyField(const Verifier &verifier, voffset_t field,
        ^~~~~~~~~~~
/data/flatbuffers/include/flatbuffers/table.h:115:8: note:   template argument deduction/substitution failed:
In file included from /data/tensorflow/tensorflow/lite/util.cc:30:
/data/tensorflow/tensorflow/lite/schema/schema_generated.h:3241:58: note:   candidate expects 3 arguments, 2 provided
            VerifyField<uint8_t>(verifier, VT_DETAILS_TYPE) &&
In file included from /data/flatbuffers/include/flatbuffers/flatbuffer_builder.h:32,
                 from /data/flatbuffers/include/flatbuffers/flatbuffers.h:27,
                 from /data/tensorflow/tensorflow/lite/schema/schema_generated.h:21,
                 from /data/tensorflow/tensorflow/lite/util.cc:30:
/data/flatbuffers/include/flatbuffers/table.h:115:8: note: candidate: 'template<class T> bool flatbuffers::Table::VerifyField(const flatbuffers::Verifier&, flatbuffers::voffset_t, size_t) const'
   bool VerifyField(const Verifier &verifier, voffset_t field,
```

hjonnala, 2022-05-17:
Hello @liviolima80 can you please try to cross compile with bazel+docker: https://github.com/google-coral/libedgetpu#bazel--docker

stiggy87, 2023-05-18:
@hjonnala I've been running into the same thing. It works fine in the bazel+docker flow, but when I try to build it in a arm64v8/ubuntu:latest image, I get similar errors.

For me, I'm using TF 2.12, and there might be a chance that the debian/ubuntu package used may be too old.

feranick, 2024-01-31:
Updated builds and repo available here:

https://github.com/feranick/libedgetpu