google-coral/libedgetpu #41: Unknown: USB transfer error 1 [LibUsbDataInCallback]  if process is kept running after inference.

waynieack, 2022-12-12:
### Description

I have a Ubuntu 20.04.5 LTS VM with a full USB 3.0 card passed through to it with nothing but the Coral USB plugged into the card. I am able to load a model and run as many inference's with as many different images as I want with no issue, but if I try to hold the process in order to keep the model loaded so I can run more  inference's later without loading the model again, the process always dies with the same Unknown: USB transfer error 1 error at the same exact spot in the debug logs. 

I don't think this has anything to do with the cable or heat as it happens in the same exact point in the debug logs and I can reproduce it in the with the same exact process every time. 

I made a very simple way to reproduce this by adding a time.sleep(300) at the bottom of the examples/classify_image.py example script in the pycoral repo.  See the full debug below:

### Relevant Log Output

```shell
/coral/pycoral# sudo -u www-data python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg
I tflite/edgetpu_manager_direct.cc:453] No matching device is already opened for shared ownership.
I driver/driver_factory_default.cc:31] Failed to open /sys/class/apex: No such file or directory
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x1a6e, product:0x89a
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[1]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[8] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[7] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[6] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[9] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[10] port[2]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[10] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[5] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[4] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[0]
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x18d1, product:0x9302
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[1]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[8] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[7] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[6] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[9] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[10] port[2]
I driver/usb/local_usb_device.cc:998] EnumerateDevices: found [/sys/bus/usb/devices/10-2]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[10] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[5] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[4] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[0]
I driver/beagle/beagle_usb_driver_provider.cc:225] Enumerate: adding path [/sys/bus/usb/devices/10-2]
I tflite/edgetpu_manager_direct.cc:471] No device of type Apex (PCIe) is available.
I tflite/edgetpu_context_direct.cc:106] USB always DFU: False (default)
I tflite/edgetpu_context_direct.cc:147] USB bulk-in queue capacity: 8
I tflite/edgetpu_context_direct.cc:67] Performance expectation: Max (default)
I driver/usb/usb_driver.cc:1383] Open device and check if DFU is needed
I driver/usb/local_usb_device.cc:1013] OpenDevice: [/sys/bus/usb/devices/10-2]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[1]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[2] port[0]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[8] port[0]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[7] port[0]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[6] port[0]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[9] port[0]
I driver/usb/local_usb_device.cc:1050] OpenDevice: checking bus[10] port[2]
I driver/usb/local_usb_device.cc:1081] OpenDevice: device opened 0x2d7a500
I driver/usb/local_usb_device.cc:184] LocalUsbDevice
I driver/usb/usb_standard_commands.cc:36] UsbStandardCommands
I driver/usb/usb_dfu_commands.cc:37] UsbDfuCommands
I driver/usb/usb_standard_commands.cc:43] GetDeviceDescriptor
I driver/usb/local_usb_device.cc:398] GetDescriptor
I driver/usb/usb_standard_commands.cc:78] Vender ID: 0x18d1
I driver/usb/usb_standard_commands.cc:79] Product ID: 0x9302
I driver/usb/usb_driver.cc:1410] Device is already in application mode, skipping DFU
I driver/usb/usb_driver.cc:1422] Resetting device
I driver/usb/local_usb_device.cc:243] Close: closing device 0x2d7a500
I driver/usb/local_usb_device.cc:216] DoCancelAllTransfers: cancelling 0 async transfers
I driver/usb/local_usb_device.cc:224] DoCancelAllTransfers: waiting for all async transfers to complete
I driver/usb/local_usb_device.cc:234] DoCancelAllTransfers: all async transfers have completed
I driver/usb/local_usb_device.cc:276] Close: releasing 0 transfer buffers
[... 604 lines cut ...]
I driver/usb/usb_driver.cc:1260] WorkerThreadFunc Installing bulk-in reader. buffer index [4]
I driver/usb/local_usb_device.cc:748] AsyncBulkInTransfer
I driver/usb/local_usb_device.cc:761] ASYNC IN 1 begin
I driver/usb/usb_driver.cc:1260] WorkerThreadFunc Installing bulk-in reader. buffer index [5]
I driver/usb/local_usb_device.cc:748] AsyncBulkInTransfer
I driver/usb/local_usb_device.cc:761] ASYNC IN 1 begin
I driver/usb/usb_driver.cc:1260] WorkerThreadFunc Installing bulk-in reader. buffer index [6]
I driver/usb/local_usb_device.cc:748] AsyncBulkInTransfer
I driver/usb/local_usb_device.cc:761] ASYNC IN 1 begin
I driver/usb/usb_driver.cc:1324] WorkerThreadFunc callback event available. skip waiting
I driver/usb/usb_driver.cc:1174] WorkerThreadFunc dispatching 1 callback events in worker thread
I driver/usb/usb_driver.cc:1260] WorkerThreadFunc Installing bulk-in reader. buffer index [7]
I driver/usb/local_usb_device.cc:748] AsyncBulkInTransfer
I driver/usb/local_usb_device.cc:761] ASYNC IN 1 begin
I driver/usb/usb_driver.cc:1317] WorkerThreadFunc waiting on state change
I driver/usb/usb_driver.cc:91] Unlocks both mutex
I driver/usb/local_usb_device.cc:672] ASYNC IN 1 end
I driver/usb/local_usb_device.cc:97] ConvertLibUsbTransferStatus: USB transfer error 1 [LibUsbDataInCallback]
I driver/usb/local_usb_device.cc:642] UnregisterCompletedTransfer
I driver/usb/usb_driver.cc:86] lock (does nothing)
I driver/usb/usb_driver.cc:1322] WorkerThreadFunc driver state change detected
I driver/usb/usb_driver.cc:1174] WorkerThreadFunc dispatching 1 callback events in worker thread
F driver/usb/usb_driver.cc:1148] HandleQueuedBulkIn transfer in failed. Unknown: USB transfer error 1 [LibUsbDataInCallback]
Aborted
```

waynieack, 2022-12-12:
Kernel Info: 5.15.0-56-generic #62~20.04.1-Ubuntu SMP

waynieack, 2022-12-12:
Also as long as I don't run an inference, I can load the model and sleep the script for as long as I want without the error happening. As soon as a inference is run and the sleep starts, it will crash with the same error every time. If I run the inference in a loop, as long as they are running, it will not crash. 

This is the script with the sleeps before and after the inference.
[https://gist.github.com/waynieack/b7002a89b49e72c3af0947e778509f0f](url)

This is the logs from running the script above.
[https://gist.github.com/waynieack/1a4e74cfa4d9811ab8ab4bc639cc1f75](url)

hjonnala, 2022-12-12:
The issue is not happenning on my Linux machine. Are you running on a virtual machine? If so, please try the same thing on native OS.

```
:uname -a
Linux hemanth 5.19.11-1rodete1-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.19.11-1rodete1 (2022-10-31) x86_64 GNU/Linux
```

waynieack, 2022-12-12:
I don't have a physical machine with the same hardware as the host, in addition the kernel version is different.  So even if it works on the physical machine, it doesn't tell me much because there's a bunch of other differences.

Regardless, I need it to work on the virtual machine not a physical machine. The issue doesn't seem to be related to throughput because its not even processing anything. I am passing through the full USB 3.0 PCI card per what I've read in the forums.

waynieack, 2022-12-13:
I moved it to a port on an onboard controller and just passed the port through and I don't have the issue anymore. I tried the card with just port pass through instead of full PCI pass through and I have the same issue. 

So it seems that there is and issue with the Coral USB and the ASM1042 USB 3.0 PCI card. The USB port pass through is not as fast as the PCI pass through but I really don't know what card to get, I have never had a device that was so picky about a USB PCI card.

hjonnala, 2022-12-14:
can you please share the inference time results for USB port pass through vs USB PCI card pass through. Thanks!!

waynieack, 2022-12-14:
To give some background on what I am doing, I am processing images from security camera footage to check for objects. When an event is recorded, the code selects a set of images and runs it through object detection. Currently it loads the model for each event and processes the images and exits, but I want to hold the model loaded (to reduce the processing time) and submit the images via an HTTP API and that is where the problem is because it crashes as we have already discussed above.  

After running with the USB port passed through for a while with the model held loaded, it seemed to have just locked up at some point. I didn't get any logs for what happened at the time because there were way to many logs. So now I am back to the USB PCI card pass through loading the model on every run. I have seen it lock up when you interrupt an inference before its finished and it requires that you unplug the Coral and plug it back in (I have tried some USB reset script which didn't seem to work).

Both of these tests were done with the same group of images. The model is loaded before each set and inference is run on each image 1 time. 

**USB Port Passthrough:**
[perf: processor:tpu Coral TPU detection took: 33.12 ms]
[perf: processor:tpu Coral TPU detection took: 33.97 ms]
[perf: processor:tpu Coral TPU detection took: 33.59 ms]
[perf: processor:tpu Coral TPU detection took: 32.74 ms]
[perf: processor:tpu Coral TPU detection took: 32.63 ms]
[perf: processor:tpu Coral TPU detection took: 33.22 ms]
[perf: processor:tpu Coral TPU detection took: 34.02 ms]
[perf: processor:tpu Coral TPU detection took: 36.11 ms]
[perf: processor:tpu Coral TPU detection took: 32.67 ms]
[perf: processor:tpu Coral TPU detection took: 33.74 ms]
[perf: processor:tpu Coral TPU detection took: 33.59 ms]
[perf: processor:tpu Coral TPU detection took: 33.50 ms]
[perf: processor:tpu Coral TPU detection took: 34.33 ms]
[perf: processor:tpu Coral TPU detection took: 34.20 ms]

**USB PCI Card Passthrough:**
[perf: processor:tpu Coral TPU detection took: 21.93 ms]
[perf: processor:tpu Coral TPU detection took: 21.76 ms]
[perf: processor:tpu Coral TPU detection took: 22.14 ms]
[perf: processor:tpu Coral TPU detection took: 22.38 ms]
[perf: processor:tpu Coral TPU detection took: 21.06 ms]
[perf: processor:tpu Coral TPU detection took: 21.33 ms]
[perf: processor:tpu Coral TPU detection took: 21.58 ms]
[perf: processor:tpu Coral TPU detection took: 22.15 ms]
[perf: processor:tpu Coral TPU detection took: 20.96 ms]
[perf: processor:tpu Coral TPU detection took: 20.90 ms]
[perf: processor:tpu Coral TPU detection took: 21.58 ms]
[perf: processor:tpu Coral TPU detection took: 23.53 ms]
[perf: processor:tpu Coral TPU detection took: 21.64 ms]
[perf: processor:tpu Coral TPU detection took: 21.19 ms]

Using the same model as above I also did the test with the examples/classify_image.py:

**USB Port Passthrough:**
/coral/pycoral# python3 examples/classify_image.py --model /var/lib/zmeventnotification/models/coral_edgetpu/ssdlite_mobiledet_coco_qat_postprocess_edgetpu.tflite --labels /var/lib/zmeventnotification/models/coral_edgetpu/coco_indexed.names --input test_data/parrot.jpg
----INFERENCE TIME----
Note: The first inference on Edge TPU is slow because it includes loading the model into Edge TPU memory.
43.6ms
27.5ms
28.5ms
29.5ms
27.7ms
-------RESULTS--------
207: 1.01903

**USB PCI Card Passthrough:**
/coral/pycoral# python3 examples/classify_image.py --model /var/lib/zmeventnotification/models/coral_edgetpu/ssdlite_mobiledet_coco_qat_postprocess_edgetpu.tflite --labels /var/lib/zmeventnotification/models/coral_edgetpu/coco_indexed.names --input test_data/parrot.jpg
Note: The first inference on Edge TPU is slow because it includes loading the model into Edge TPU memory.
30.0ms
14.0ms
14.1ms
14.2ms
13.8ms
-------RESULTS--------
207: 1.01903