google-coral/libedgetpu #45: macOS Ventura did not compile ... make failed

Smiril, 2023-04-19:
### Description

hello some one linked me to this repo ...

well installed decencies and build failed

### Issue Type

Build/Install

### Operating System

Mac OS

### Coral Device

USB Accelerator

### Programming Language

C++, Python 3.7, Python 3.9, Other

### Relevant Log Output

```shell
Starting local Bazel server and connecting to it...
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
 checking cached actions
Loading: 0 packages loaded
Loading: 1 packages loaded
bazel build --stripopt=-x --compilation_mode=opt --cpu=darwin_arm64 --embed_label='TENSORFLOW_COMMIT=a4dfb8d1a71385bd6d122e4f27f86dcebb96712d' --stamp //tflite/public:libedgetpu_direct_usb.dylib.stripped
ERROR: /private/var/tmp/_bazel_smiril/88e6a2f750982e644b391931249c9061/external/bazel_tools/platforms/BUILD:34:6: in alias rule @bazel_tools//platforms:aarch64: Constraints from @bazel_tools//platforms have been removed. Please use constraints from @platforms repository embedded in Bazel, or preferably declare dependency on https://github.com/bazelbuild/platforms. See https://github.com/bazelbuild/bazel/issues/8622 for details.
ERROR: /private/var/tmp/_bazel_smiril/88e6a2f750982e644b391931249c9061/external/bazel_tools/platforms/BUILD:34:6: Analysis of target '@bazel_tools//platforms:aarch64' failed
ERROR: /Users/smiril/coral/libedgetpu/tflite/public/BUILD:142:10: While resolving toolchains for target //tflite/public:libedgetpu_direct_usb.dylib: Target @local_execution_config_platform//:platform was referenced as a platform, but does not provide PlatformInfo
ERROR: Analysis of target '//tflite/public:libedgetpu_direct_usb.dylib.stripped' failed; build aborted: 
INFO: Elapsed time: 0,374s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (7 packages loaded, 9 targets conf\
igured)
make: *** [libedgetpu-direct] Error 1
```

Smiril, 2023-04-19:
well need the .dylib for x86_64 and arm64 bundled as universal lib

hjonnala, 2023-04-19:
please try with bazel version 4.2.2: https://github.com/google-coral/libedgetpu/blob/master/docker/Dockerfile#L57

Smiril, 2023-04-20:
i'm on a arm64 system is there a arm64 installer?

Smiril, 2023-04-20:
mark as solved

I download https://coral.ai/software the binary

and 
```
cd edgetpu_runtime/libedgetpu/direct/
mkdir macos
lipo darwin_arm64/libedgetpu.1.0.dylib darwin_x86_64/libedgetpu.1.0.dylib -output macos/libedgetpu.1.0.dylib -create
```

closed 2023-04-20 by Smiril