google-coral/libedgetpu #48: M.2 TPU device violates PCI specification

lamw, 2023-08-18:
### Description

Customers that attempt to passthrough the M.2 TPU to a Virtual Machine using VMware ESXi Hypervisor have found that the Apex driver fails to initialize.

```
# dmesg
<snip>
[    3.780139] apex 0000:02:03.0: enabling device (0000 -> 0002)
[    3.785860] apex 0000:02:03.0: Page table init timed out
[    3.786103] apex 0000:02:03.0: MSI-X table init timed out
```

Upon initial investigation from VMware Engineering, the following was concluded:

```
Unfortunately the device in question violates PCI specification by mapping PBA, MSI-X vector table, and other registers into same 4KB page (PBA is at 0x46068, VT at 0x46800, but there is a bunch of other registers in 0x46XXX range). PCIe spec 6.0, page 1020, has this to say:

<quote>
If a Base Address Register or entry in the Enhanced Allocation capability that maps address space for the MSI-X Table or
MSI-X PBA also maps other usable address space that is not associated with MSI-X structures, locations (e.g., for CSRs)
used in the other address space must not share any naturally aligned 4-KB address range with one where either MSI-X
structure resides. This allows system software where applicable to use different processor attributes for MSI-X structures
and the other address space. (Some processor architectures do not support having different processor attributes
associated with the same naturally aligned 4-KB physical address range.) The MSI-X Table and MSI-X PBA are permitted
to co-reside within a naturally aligned 4-KB address range, though they must not overlap with each other.
</quote>

So having CSR registers in same page as MSI-X VT page violates the spec, and under ESXi CSR registers become unreachable (writes ignored, reads return zeroes). Due to this device driver cannot correctly initialize device.

If firmware can modify device's behavior so that VT/PBA arrays do not share same 4KB page with other registers, device will work with ESXi's passthrough. Or if firmware can hide MSI-X capability from PCI configuration space, that would fix issue as well.
```

I'm not sure if this has already been reported but if Google/Coral can either fix the behavior of the device to conform to the PCI specification OR hide MSI-X capability, then successful passthrough of the M.2 TPU should function correctly when using ESXi, which is a popular Hypervisor platform for development purpose

### Issue Type

Build/Install

### Operating System

Ubuntu

### Coral Device

M.2 Accelerator A+E

goldserve, 2023-10-20:
Yes, please do look into addressing this!

ManuelPerrot, 2023-10-20:
Very interested to have this fixed as well.
Looks like Xen could have the same issue: https://xcp-ng.org/forum/topic/6304/google-coral-tpu-pcie-passthrough-woes/20

k1n6b0b, 2023-10-20:
Adding another vote to fix this here!! There are a ton of threads/requests for this but they're all directed at frigate and VMware instead of here.

grembling22, 2023-10-31:
+1 for a fix

tbozik, 2023-11-06:
+1 for a fix

kentkravitz, 2023-11-11:
+1 fix please.

TokugawaHeavyIndustries, 2023-11-13:
+1 for fix, commenting to follow

kentkravitz, 2023-11-15:
Can anyone think of any other possible workarounds for this problem?  Seems like ESXi could also use a quirks mode for pci-e cards that need some tweaking.

kuantek, 2023-11-19:
+1 for a fix please

Brandon314, 2023-11-24:
+1 for a fix please

gknepper, 2023-11-25:
+1 for a fix please

vobelic, 2024-01-06:
+1 for the fix

zaolin, 2024-01-19:
Just try to disable the msi bus for the bridge if possible, echo 1 > /sys/bus/pci/devices/$bridge/msi_bus

zaolin, 2024-01-19:
Or just patch out msi-x of the driver: https://github.com/google/gasket-driver/blob/09385d485812088e04a98a6e1227bf92663e0b59/src/gasket_interrupt.c#L245

I am quoting:

/*
 * On QCM DragonBoard, we exit gasket_interrupt_msix_init() and kernel interrupt
 * setup code with MSIX vectors masked. This is wrong because nothing else in
 * the driver will normally touch the MSIX vectors.
 *
 * As a temporary hack, force unmasking there.
 *
 * TODO: Figure out why QCM kernel doesn't unmask the MSIX vectors, after
 * gasket_interrupt_msix_init(), and remove this code.
 */

bridge-four, 2024-01-22:
+1 vote for fix!

alexsahka, 2024-01-28:
+1 vote for fix!

thefl0yd, 2024-02-10:
This is not likely to ever get fixed now with broadcom deprecating free ESXi. Aware this is a TPU issue but the ESXi userbase is just going to keep shrinking at this point.

SunvidWong, 2024-06-07:
+1 vote for fix!

johnlento, 2024-10-04:
> Just try to disable the msi bus for the bridge if possible, echo 1 > /sys/bus/pci/devices/$bridge/msi_bus as a temporary fix. For me it looks like there is a lot of hacky stuff in the kernel driver: https://github.com/google/gasket-driver/blob/09385d485812088e04a98a6e1227bf92663e0b59/src/gasket_interrupt.c#L245

This already appears to be set on my fresh install and passthrough. No luck.

obsidiangroup, 2025-02-16:
Feels like even a year later, no fix or comments on this. We also had a need to deploy these in the field in an enterprise setting. Now we have to decide to either use a bare metal host or use USB, which is less than ideal.

thefl0yd, 2025-02-16:
> Feels like even a year later, no fix or comments on this. We also had a need to deploy these in the field in an enterprise setting. Now we have to decide to either use a bare metal host or use USB, which is less than ideal.

As I said almost a year ago - this is not going to get fixed. The coral is google abandonware and ESXi is a lot less interesting to a lot of people given the direction broadcom has taken the product.

johnlento, 2025-02-16:
Pretty sure the Google Coral team doesn’t exist anymore. It’s probably like one PM.

Sent via RFC 1149
Get some Carrier Pigeons<https://en.m.wikipedia.org/wiki/IP_over_Avian_Carriers>...

DemiMarie, 2025-05-12:
This should work fine with Xen.