google-coral/libedgetpu #51: Windows 11 USB Accelerator USB error -99 [OpenDevice]

JorgeRuizITCL, 2023-11-22:
### Description

Hi,

I'm encountering issues with the Coral Edge TPU USB 3.0 Accelerator on Windows 11, specifically related to poor performance  and stability. In this case I am using [EfficientDet-Lite0 (int8)](https://storage.googleapis.com/mediapipe-models/object_detector/efficientdet_lite0/int8/latest/efficientdet_lite0.tflite) along with pycoral version 2.0.0.

When using the **edgetpu_runtime_20221024 performance** driver (latest), the accelerator has a throughput of 30fps but after exiting the inference loop, running the program for the second time pycoral fails to use the accelerator until a system restart or a port change.

Using the **edgetpu_runtime_20210119 performance** driver, the accelerator performance drops to 15fps, but the usb seems to be released every time, so the script can be run as many times as needed. 

### Issue Type

Performance

### Operating System

Windows 10

### Coral Device

USB Accelerator

### Programming Language

Python 3.9

### Relevant Log Output

```shell
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x1a6e, product:0x89a
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[11]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[6]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[12]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[13]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[25]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[1]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[3]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[4]
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x18d1, product:0x9302
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[11]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[6]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[12]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[13]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[25]
I driver/usb/local_usb_device.cc:998] EnumerateDevices: found [/sys/bus/usb/devices/1-25]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[1]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[3]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[4]
I driver/beagle/beagle_usb_driver_provider.cc:225] Enumerate: adding path [/sys/bus/usb/devices/1-25]
I driver/driver_factory_windows.cc:29] DriverFactoryWin::EnumerateByClass()...
I driver/driver_factory_windows.cc:75] DriverFactoryWin::EnumerateByClass returns empty list.
[ INFO ] 10:53:51.397 init [{'type': 'usb', 'path': '/sys/bus/usb/devices/1-25'}]
I tflite/edgetpu_manager_direct.cc:453] No matching device is already opened for shared ownership.
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x1a6e, product:0x89a
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[11]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[6]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[12]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[13]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[25]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[1]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[3]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[4]
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x18d1, product:0x9302
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[11]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[6]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[12]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[13]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[25]
I driver/usb/local_usb_device.cc:998] EnumerateDevices: found [/sys/bus/usb/devices/1-25]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[3] port[1]
[... 187 lines cut ...]
  File "C:\Users\<ME>\Documents\coral\.venv\lib\site-packages\tflite_runtime\interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "C:\Users\<ME>\Documents\coral\.venv\lib\site-packages\tflite_runtime\interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\<ME>\Documents\coral\main.py", line 70, in <module>
    main()
  File "C:\Users\<ME>\Documents\coral\main.py", line 66, in main
    engine.infer(media)
  File "C:\Users\<ME>\Documents\coral\src\core\inference.py", line 287, in infer
    human_ie.init(self._human.model, self._on_yolo)
  File "C:\Users\<ME>\Documents\coral\src\ie\abstract_tflite.py", line 21, in init
    self._interpreter = inter = tpu.make_interpreter(self._model_path, self._device)
  File "C:\Users\<ME>\Documents\coral\.venv\lib\site-packages\pycoral\utils\edgetpu.py", line 87, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "C:\Users\<ME>\Documents\coral\.venv\lib\site-packages\pycoral\utils\edgetpu.py", line 52, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "C:\Users\<ME>\Documents\coral\.venv\lib\site-packages\tflite_runtime\interpreter.py", line 162, in load_delegate
    raise ValueError('Failed to load delegate from {}\n{}'.format(
ValueError: Failed to load delegate from edgetpu.dll
```

pooltrust, 2024-06-19:
Hi, 
did you find the problem? Because I am running into the same problem and don't know how to fix it. 
Cheers

JorgeRuizITCL, 2024-06-19:
Nope :|

pooltrust, 2024-06-19:
ok, but thanks anyway for the fast response!