google-coral/libedgetpu #53: Support for phyton 3.11 and modern versions of TensorFlow is broken

feranick, 2023-12-13:
### Description

tflite-runtime (which is needed for any high level API with edgetpu) is available only with support for Python 3.9. Most distribution now no longer ship with that version, rather either 3.10 or 3.11. So there is factually no version of tflite-runtime available to download with support for coral edgetpu. The only versions of TF that work are 2.11.1 and earlier. Later versions up to the current TF 2.15.0 leads to a segfault. While this al points to a lack of maintenance, it would be great to at least update so it can be used with, well, pretty much any modern version of OS and devices. 

### Issue Type

Bug, Build/Install, Support

### Operating System

Linux, Ubuntu, Mac OS

### Coral Device

USB Accelerator, Mini PCIe, M.2 Accelerator A+E, M.2 Accelerator B+M, M.2 Accelerator with dual Edge TPU

### Other Devices

Raspberry Pi 3, Rapsberry Pi 4

### Programming Language

Other

Smiril, 2023-12-20:
I’d like Debian and macOS
With macOS i got 2 usb Coral Edge Tour
And on Debian got one pcie dual edge tpu A+E

Non will work with updated systems 
Minimum python version 3.11 on both systems

Outdated systems are a security break while programming and possible hardware leaks with memory

Well please update all libs and parts to python 3.11

stakach, 2023-12-25:
I could only get v2.8.4 and below working
https://github.com/spider-gazelle/edge-ai/issues/2

hardware is still shipping so I would assume it should still be supported?

feranick, 2023-12-25:
I managed to compile from source an operational version of tflite_runtime v2.11.1 (nothing special about the builds, just followed instructions). Wheels here:

[cp310_linux_x86_64
](https://www.dropbox.com/scl/fi/uyif5p9g1pgn85e4u80wo/tflite_runtime-2.11.1-cp310-cp310-linux_x86_64.whl?rlkey=r7am6q6k854k0snqatde14wju&dl=0)

feranick, 2023-12-25:
Note that Jamboard will be discontinued by the end of 2024.

https://support.google.com/a/answer/13342662?sjid=205795280203553201-NA

feranick, 2024-02-29:
Thanks very much to @pkgoogle and @Namburger at Google for merging PR. The libedgetpu library is now fully updated, and I hope binaries will be made available soon through the official channel. I am going to close this issue.

closed 2024-02-29 by feranick