google-coral/libedgetpu #55: Can't use Makefile build

drosseau, 2023-12-24:
### Description

I'm trying to build this library with the Makefile method, but I can't seem to get it working.

Here is my process:

```
CODEPATH=/home/user/code/github.com
EDGETPU=$CODEPATH/google-coral/libedgetpu
TFROOT=$CODEPATH/tensorflow/tensorflow

$ git clone https://github.com/tensorflow/tensorflow $TFROOT
$ git clone https://github.com/google-coral/libedgetpu $EDGETPU
$ cd $TFROOT
$ git checkout v2.11.1
$ cd $EDGETPU
$ git checkout release-grouper
make SHELL=$(which bash) TFROOT=$TFROOT -f makefile_build/Makefile libedgetpu
make: *** No rule to make target '/home/user/code/github.com/tensorflow/tensorflow/tensorflow/lite/c/common.c', needed by '/home/user/code/github.com/google-coral/libedgetpu/makefile_build/../out//home/user/code/github.com/tensorflow/tensorflow/tensorflow/lite/c/common.o'.  Stop.
```

Sadly I'm not too familiar with debugging Makefiles so I'm not able to fully understand why that error is occuring

### Issue Type

Build/Install

### Operating System

Linux

### Coral Device

USB Accelerator

tosa-no-onchan, 2023-12-29:
I got same error too.
I think, libedgetpu is suitable for tensorflow 2.5.
Try it with tensorflow 2.5.
See https://github.com/google-coral/libedgetpu/blob/master/workspace.bzl

$ git clone https://github.com/tensorflow/tensorflow
$ cd tensorflow
$ git checkout a4dfb8d1a71385bd6d122e4f27f86dcebb96712d -b tf2.5

tosa-no-onchan, 2023-12-29:
I found the flowwing,

in libedgetpu/makefile_build/Makefile
...
65 LIBEDGETPU_CSRCS := $(TFROOT)/tensorflow/lite/c/common.c
66 LIBEDGETPU_COBJS := $(call TOBUILDDIR,$(patsubst %.c,%.o,$(LIBEDGETPU_CSRCS)))

But  in tensorflow newer verion than 2.5,
there is not  common.c on  $(TFROOT)/tensorflow/lite/c/

tosa-no-onchan, 2023-12-29:
edit  libedgetpu/makefile_build/Makefile
...
65 #LIBEDGETPU_CSRCS := $(TFROOT)/tensorflow/lite/c/common.c
65 LIBEDGETPU_CSRCS := $(TFROOT)/tensorflow/lite/core/c/common.cc
...
187 #$(LIBEDGETPU_COBJS) : $(BUILDDIR)/%.o: %.c
187 $(LIBEDGETPU_COBJS) : $(BUILDDIR)/%.o: %.cc
...

and try it with tf 2.11.1

tosa-no-onchan, 2023-12-29:
Sorry,
edit libedgetpu/makefile_build/Makefile
does not work.
try with tf2.5

closed 2024-01-02 by drosseau