google-coral/libedgetpu #58: Build is failing

dandurisrihari, 2024-01-24:
### Description

  404  Not Found
Ign:15 http://deb.debian.org/debian stretch-updates/main arm64 Packages
Reading package lists...
W: The repository 'http://security.debian.org/debian-security stretch/updates Release' does not have a Release file.
W: The repository 'http://deb.debian.org/debian stretch Release' does not have a Release file.
W: The repository 'http://deb.debian.org/debian stretch-updates Release' does not have a Release file.
E: Failed to fetch http://security.debian.org/debian-security/dists/stretch/updates/main/binary-armhf/Packages  404  Not Found [IP: 151.101.2.132 80]
E: Failed to fetch http://deb.debian.org/debian/dists/stretch/main/binary-armhf/Packages  404  Not Found
E: Failed to fetch http://deb.debian.org/debian/dists/stretch-updates/main/binary-armhf/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.

### Issue Type

Build/Install

### Operating System

Ubuntu

feranick, 2024-01-31:
Updated builds and repo available here:

https://github.com/feranick/libedgetpu

feranick, 2024-02-29:
The official repo has been updated. Please try again, should work now.

ARusDian, 2024-07-17:
got similiar error code on tensorflow 17

` DOCKER_CPUS="k8" DOCKER_IMAGE="ubuntu:22.04" DOCKER_TARGETS=libedgetpu make docker-build
docker build  -t "libedgetpu-cross-ubuntu-22.04" \
    --build-arg IMAGE=ubuntu:22.04 /home/dian/libedgetpu/docker
Sending build context to Docker daemon  8.704kB
Step 1/10 : ARG IMAGE
Step 2/10 : FROM ${IMAGE}
 ---> d04dcc2ab57b
Step 3/10 : COPY update_sources.sh /
 ---> Using cache
 ---> 578e71103861
Step 4/10 : RUN /update_sources.sh
 ---> Using cache
 ---> f60b0fec6e7c
Step 5/10 : RUN dpkg --add-architecture armhf
 ---> Using cache
 ---> a7f8509a581a
Step 6/10 : RUN dpkg --add-architecture arm64
 ---> Using cache
 ---> f13b3733af21
Step 7/10 : RUN apt-get update     && DEBIAN_FRONTEND=noninteractive apt-get install -y       libc6-dev:arm64       libc6-dev:armhf     && DEBIAN_FRONTEND=noninteractive apt-get install -y       python3-all       python3-numpy       build-essential       crossbuild-essential-armhf       crossbuild-essential-arm64       libusb-1.0-0-dev       libusb-1.0-0-dev:arm64       libusb-1.0-0-dev:armhf       zlib1g-dev       zlib1g-dev:armhf       zlib1g-dev:arm64       sudo       debhelper       pkg-config       zip       unzip       curl       wget       git       tree       software-properties-common       $(grep Ubuntu /etc/os-release > /dev/null && echo vim-common || echo xxd)
 ---> Running in be58623564c0
Get:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease [270 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease [128 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease [127 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease [129 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports jammy/universe arm64 Packages [17.2 MB]
Ign:6 http://ports.ubuntu.com/ubuntu-ports jammy/restricted amd64 Packages
Get:7 http://ports.ubuntu.com/ubuntu-ports jammy/main armhf Packages [1701 kB]
Ign:8 http://ports.ubuntu.com/ubuntu-ports jammy/universe amd64 Packages
Ign:9 http://ports.ubuntu.com/ubuntu-ports jammy/main amd64 Packages
Ign:10 http://ports.ubuntu.com/ubuntu-ports jammy/multiverse amd64 Packages
Get:11 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 Packages [1758 kB]
Get:12 http://ports.ubuntu.com/ubuntu-ports jammy/universe armhf Packages [16.8 MB]
Get:13 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 Packages [1991 kB]
Get:14 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe armhf Packages [1074 kB]
Ign:15 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main amd64 Packages
Get:16 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main armhf Packages [1134 kB]
Ign:17 http://ports.ubuntu.com/ubuntu-ports jammy-updates/multiverse amd64 Packages
Get:18 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe arm64 Packages [1343 kB]
Ign:19 http://ports.ubuntu.com/ubuntu-ports jammy-updates/restricted amd64 Packages
Ign:20 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe amd64 Packages
Ign:21 http://ports.ubuntu.com/ubuntu-ports jammy-backports/universe amd64 Packages
Ign:22 http://ports.ubuntu.com/ubuntu-ports jammy-backports/main amd64 Packages
Ign:23 http://ports.ubuntu.com/ubuntu-ports jammy-security/multiverse amd64 Packages
Ign:24 http://ports.ubuntu.com/ubuntu-ports jammy-security/restricted amd64 Packages
Ign:25 http://ports.ubuntu.com/ubuntu-ports jammy-security/universe amd64 Packages
Ign:26 http://ports.ubuntu.com/ubuntu-ports jammy-security/main amd64 Packages
Get:27 http://ports.ubuntu.com/ubuntu-ports jammy-security/universe armhf Packages [801 kB]
Get:28 http://ports.ubuntu.com/ubuntu-ports jammy-security/main arm64 Packages [1731 kB]
Get:29 http://ports.ubuntu.com/ubuntu-ports jammy-security/main armhf Packages [884 kB]
Get:30 http://ports.ubuntu.com/ubuntu-ports jammy-security/universe arm64 Packages [1065 kB]
Ign:6 http://ports.ubuntu.com/ubuntu-ports jammy/restricted amd64 Packages
Ign:8 http://ports.ubuntu.com/ubuntu-ports jammy/universe amd64 Packages
Ign:9 http://ports.ubuntu.com/ubuntu-ports jammy/main amd64 Packages
Ign:10 http://ports.ubuntu.com/ubuntu-ports jammy/multiverse amd64 Packages
Ign:15 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main amd64 Packages
Ign:17 http://ports.ubuntu.com/ubuntu-ports jammy-updates/multiverse amd64 Packages
Ign:19 http://ports.ubuntu.com/ubuntu-ports jammy-updates/restricted amd64 Packages
Ign:20 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe amd64 Packages
Ign:21 http://ports.ubuntu.com/ubuntu-ports jammy-backports/universe amd64 Packages
Ign:22 http://ports.ubuntu.com/ubuntu-ports jammy-backports/main amd64 Packages
Ign:23 http://ports.ubuntu.com/ubuntu-ports jammy-security/multiverse amd64 Packages
Ign:24 http://ports.ubuntu.com/ubuntu-ports jammy-security/restricted amd64 Packages
Ign:25 http://ports.ubuntu.com/ubuntu-ports jammy-security/universe amd64 Packages
Ign:26 http://ports.ubuntu.com/ubuntu-ports jammy-security/main amd64 Packages
Ign:6 http://ports.ubuntu.com/ubuntu-ports jammy/restricted amd64 Packages
Ign:8 http://ports.ubuntu.com/ubuntu-ports jammy/universe amd64 Packages
Ign:9 http://ports.ubuntu.com/ubuntu-ports jammy/main amd64 Packages
[... 54 lines cut ...]
Ign:24 http://ports.ubuntu.com/ubuntu-ports jammy-security/restricted amd64 Packages
Ign:25 http://ports.ubuntu.com/ubuntu-ports jammy-security/universe amd64 Packages
Ign:26 http://ports.ubuntu.com/ubuntu-ports jammy-security/main amd64 Packages
Fetched 48.1 MB in 2min 47s (289 kB/s)
[... cut ...] ild-essential-armhf       crossbuild-essential-arm64       libusb-1.0-0-dev       libusb-1.0-0-dev:arm64       libusb-1.0-0-dev:armhf       zlib1g-dev       zlib1g-dev:armhf       zlib1g-dev:arm64       sudo       debhelper       pkg-config       zip       unzip       curl       wget       git       tree       software-properties-common       $(grep Ubuntu /etc/os-release > /dev/null && echo vim-common || echo xxd)' returned a non-zero code: 100
make: *** [Makefile:179: docker-image] Error 100
`

feranick, 2024-07-17:
What are you targeting with this build? Ubuntu 22.04? Debian? Else? Most of the errors point to a problem accessing repos.

feranick, 2024-07-17:
You are compiling with the wrong architecture (k8 is x86_64, you need aarch64) and OS (debian bookworm, not ubuntu 22.04). Try:
```
DOCKER_CPUS="aarch64" DOCKER_IMAGE="debian:bookworm" DOCKER_TARGETS=libedgetpu make docker-build
```

feranick, 2024-07-17:
Or, since I am assuming you are compiling this on the Rpi directly, you can use Bazel directly, as stated on the README.md file.

feranick, 2024-07-17:
Prebuilt debs are available in my fork:

https://github.com/feranick/libedgetpu/releases/tag/16.0TF2.17.0-1

ARusDian, 2024-07-17:
> You are compiling with the wrong architecture (k8 is x86_64, you need aarch64) and OS (debian bookworm, not ubuntu 22.04). Try:
> 
> ```
> DOCKER_CPUS="aarch64" DOCKER_IMAGE="debian:bookworm" DOCKER_TARGETS=libedgetpu make docker-build
> ```

Oh yeah, sorry about that. I understand now what the README.md file means. I was mindlessly copying both commands, but I should choose one instead. Thank you! 👌

ARusDian, 2024-07-17:
got another error after install after install bazel, the error says

`/usr/local/bin/bazel: line 223: /usr/local/lib/bazel/bin/bazel-real: cannot execute binary file: Exec format error
`
here the log after step 10/10

`Step 10/10 : RUN wget -O /bazel https://github.com/bazelbuild/bazel/releases/download/${BAZEL_VERSION}/bazel-${BAZEL_VERSION}-installer-linux-x86_64.sh &&     bash /bazel &&     rm -f /bazel
 ---> Using cache
 ---> 91641c0251bd
Successfully built 91641c0251bd
Successfully tagged libedgetpu-cross-debian-bookworm:latest
docker run --rm -i --tty \
    -v /home/dian/libedgetpu:/workspace \
    "libedgetpu-cross-debian-bookworm" /bin/bash -c "chmod a+w /; groupadd --gid 1000 dian; useradd -m -e '' -s /bin/bash --gid 1000 --uid 1000 dian; echo 'dian ALL=(ALL) NOPASSWD:ALL' >> /etc/sudoers; su dian -c 'for cpu in aarch64; do make CPU=\${cpu} -C /workspace libedgetpu || exit 1; done'"
make: Entering directory '/workspace'
/usr/local/bin/bazel: line 223: /usr/local/lib/bazel/bin/bazel-real: cannot execute binary file: Exec format error
/usr/local/bin/bazel: line 223: /usr/local/lib/bazel/bin/bazel-real: Success
bazel build --stripopt=-x --compilation_mode=opt --cpu=aarch64 --embed_label='TENSORFLOW_COMMIT=' --stamp //tflite/public:libedgetpu_direct_all.so.stripped
/usr/local/bin/bazel: line 223: /usr/local/lib/bazel/bin/bazel-real: cannot execute binary file: Exec format error
/usr/local/bin/bazel: line 223: /usr/local/lib/bazel/bin/bazel-real: Success
make: *** [Makefile:99: libedgetpu-direct] Error 1
make: Leaving directory '/workspace'
make: *** [Makefile:188: docker-build] Error 1`

feranick, 2024-07-17:
Are you running it within docker?

feranick, 2024-07-17:
You are using a version of bazel that is not for aarch64. Are you compiling directly on the RPi? If so there is no version of bazel for aarch64/linux for whatever reason. 

I'd recommend you try to use a x86_64 system and run docker from there for your architecture (which is what I usually do), or use prebuilt libraries (URL above).

ARusDian, 2024-07-17:
> You are using a version of bazel that is not for aarch64. Are you compiling directly on the RPi? If so there is no version of bazel for aarch64/linux for whatever reason.

sorry for error before, i'm tried another approach by using bazel to compile directly to the RPI4
i'm still failing though when trying to build bazel for RPi4, and also i'm not sure about using docker running the libedgetpu later on rpi4 because i need to access camera (picamera2) and GPIo Soon for my project

feranick, 2024-07-17:
To be clear: you don't need docker to run the library, only to compile it. Unless you care specifically to do the compilation yourself (which may be challenging on the Pi itself), I'd recommend you install the library binaries from here (my forked repo, which basically gives you the latest binaries):

https://github.com/feranick/libedgetpu/releases/tag/16.0TF2.17.0-1

You need to download the version you need (standard or max, see official documentation, usually standard is fine) and install it;
'''
wget https://github.com/feranick/libedgetpu/releases/download/16.0TF2.17.0-1/libedgetpu1-std_16.0tf2.17.0-1.bookworm_arm64.deb
sudo dpkg -i libedgetpu1-std_16.0tf2.17.0-1.bookworm_arm64.deb
'''

ARusDian, 2024-07-17:
> To be clear: you don't need docker to run the library, only to compile it. Unless you care specifically to do the compilation yourself (which may be challenging on the Pi itself), I'd recommend you [...]

ahhh i see thanks man so it means no need to build by my own right ? and also what is the different may i ask ?

feranick, 2024-07-17:
You don't need to build your own. There is no difference in what I build and how it is indicated in the official site. 

If you are referring to max vs std, one pushes the edgetpu to the max frequency, with possible overheating, while the other has lower thermal ceilings.

ARusDian, 2024-07-17:
> You don't need to build your own. There is no difference in what I build and how it is indicated in the official site.
> 
> If you are referring to max vs std, one pushes the edgetpu to the max frequency, with possible overheating, while the other has lower thermal ceilings.

aight understandable

thank you man for your help it works fine now 
[image]

wish the best for you ❤️

ARusDian, 2024-07-17:
also 

> If that works, please feel free to close this issue.

i'm not start this issue but may i close this ?

feranick, 2024-07-17:
> i'm not start this issue but may i close this ?

Ah, sorry, I thought you started it...