google-coral/libedgetpu #65: Bazel native build not working for multiple older version of TF. Python dependency and commit hash issues.

ClayJay3, 2024-03-22:
### Description

Some of the recent changes being merged have broken bazel builds using the Makefile for some of the older builds of Tensorflow. I am currently attempting to build libedgetpu for TF 2.13.1. 

It seems that the commit hash is not getting set in the bazel build command ran by the Makefile shown here:
`bazel build --stripopt=-x --compilation_mode=opt --cpu=k8 --embed_label='TENSORFLOW_COMMIT=' --stamp //tflite/public:libedgetpu_direct_all.so.stripped`

The only modification I made to the workspace.bzl file was commenting out the TF 2.16.1 commit hashes and uncommenting TF 2.13.1 commit hashes.

Bazel also seems to be unable to pull the python toolchain for tensorflow, not sure why. Checking out a82c669fb7a9b2e813cfb3d5409fea98d6a6ac8c before the toolchains were updated fixes both the python issue and commit not being set in the bazel build command. I am also able to fully build libedgetpu for TF 2.13.1 when checked out to this commit.

### Issue Type

Build/Install

### Operating System

Linux, Ubuntu

### Coral Device

USB Accelerator, M.2 Accelerator with dual Edge TPU

### Programming Language

C++

### Relevant Log Output

```shell
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 0 packages loaded
ERROR: error loading package '': Label '@org_tensorflow//tensorflow/tools/toolchains/python:python_repo.bzl' is invalid because 'tensorflow/tools/toolchains/python' is not a package; perhaps you meant to put the colon here: '@org_tensorflow//tensorflow/tools/toolchains:python/python_repo.bzl'?
Loading: 0 packages loaded
Loading: 0 packages loaded
bazel build --stripopt=-x --compilation_mode=opt --cpu=k8 --embed_label='TENSORFLOW_COMMIT=' --stamp //tflite/public:libedgetpu_direct_all.so.stripped
Another command holds the client lock: 
pid=5283
owner=client
cwd=/tmp/libedgetpu
tty=/dev/pts/1

Waiting for it to complete...
ERROR: error loading package '': Label '@org_tensorflow//tensorflow/tools/toolchains/python:python_repo.bzl' is invalid because 'tensorflow/tools/toolchains/python' is not a package; perhaps you meant to put the colon here: '@org_tensorflow//tensorflow/tools/toolchains:python/python_repo.bzl'?
INFO: Elapsed time: 0.037s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (0 packages loaded)
Loading: 0 packages loaded
ERROR: error loading package '': Label '@org_tensorflow//tensorflow/tools/toolchains/python:python_repo.bzl' is invalid because 'tensorflow/tools/toolchains/python' is not a package; perhaps you meant to put the colon here: '@org_tensorflow//tensorflow/tools/toolchains:python/python_repo.bzl'?
Loading: 0 packages loaded
Loading: 0 packages loaded
make: *** [Makefile:99: libedgetpu-direct] Error 1
```

ClayJay3, 2024-04-01:
Bump, push

ClayJay3, 2024-04-05:
Bump, push

ClayJay3, 2024-04-05:
I'm just going to use docker I guess.

SeunghyunOh-Cochl, 2024-07-03:
Hello! Do you resolve this issue?

ClayJay3, 2024-07-03:
No, it seems development had generally stopped and support for make builds isn't great. I find the using the docker environment to compile from source works for most versions.

Even if you get the makefile build to finish, it's not guaranteed to work. I had to change versions and then build and export with docker to get a working edgetpu library.

feranick, 2024-07-16:
Try again, there have been several updates to fix the exact issues you are having. The current libesdgetpu builds against TF 2.16.1 successfully.