google-coral/libedgetpu #72: Makefile is not compatible with tf 2.16.1

mwprado, 2024-08-16:
If you try compile using make, shows there isn't  common.c file. In Tf 2.16.1 there is only common.cc .

FeeJai, 2024-09-05:
got the same issue here.

mwprado, 2024-09-18:
> got the same issue here.
I fixed makefile:

[new_makefile.patch.txt](https://github.com/user-attachments/files/17037296/new_makefile.patch.txt)

mwprado, 2024-09-18:
I think that TF-2.17.0 replace common_internal.cc for common.cc .

jagiella, 2024-09-30:
@feranick without the patch provided by @mwprado it won't compile, even with the newest commit of the main branch.

Additionally, it is important to mention, that the documentation is misleading (not to say wrong) in respect to the build dependencies: On ubuntu / debian systems the provided package for libflatbuffers-dev uses a different scheme for version numbers. The current versions are v**2**.x, but tensorflow requests v**23**.x.y. So builds will never work. 

It would be great to change the documentation and add instructions to build and install the matching flatbuffer version which tensorflow requires.

## Please add to makefile_build documentation
For example for tensorflow 2.16.1 you require to build and install libflatbuffers 23.5.26:
```bash
git clone https://github.com/google/flatbuffers.git
cd flatbuffers/
git checkout v23.5.26
mkdir build && cd build
cmake .. -DFLATBUFFERS_BUILD_SHAREDLIB=ON -DFLATBUFFERS_BUILD_TESTS=OFF -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr
make
sudo make install
```

## Please add to main documentation
And in order to make debian packages using the Makefile-only approach you need to build both versions of the library first (std, max) before the packaging routine can successfully been run:
 ```bash
TFROOT=<path to tensorflow source>/ LD_LIBRARY_PATH=/usr/local/lib/ make -f makefile_build/Makefile  libedgetpu-throttled
TFROOT=<path to tensorflow source>/ LD_LIBRARY_PATH=/usr/local/lib/ make -f makefile_build/Makefile  libedgetpu-throttled
debuild -us -uc -tc -b -a amd64 -d
```

feranick, 2024-09-30:
@jagiella I am not sure why you are tagging me on this. I am not a Google developer and have no access to the main repo. Feel free to create an actual patch, make a PR against the main repo and hope that it gets merged. 

I maintain a separate fork that is updated:
https://github.com/feranick/libedgetpu/

with a PR to the main repo:
https://github.com/google-coral/libedgetpu/pull/70