google-coral/libedgetpu #75: Failed to load delegate from libedgetpu.so.1 with M.2 Accelerator A+E key

GaryHuang-ASUS, 2024-10-22:
### Description

Description
I'm using the M.2 Accelerator A + E to do some PoC.

Currently I open the Kernel Config to build the necessary kernel module (coral tpu driver) and follow the official example to verify. I can see /dev/apex_0 after I insmod gaket & apex, but when I run the example python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt  --input test_data/parrot.jpg in the container (debian:10) it shows this error.

[image]

The error also appear in Frigate NVR. It's weird that it shows no TPU device was detected, but there's /dev/apex_0 and also libedgetpu.so.1.0 under /usr/lib/aarch-linux-gnu directory.

[image]

[image]

[image]

[image]

What do I need to do for fixing it?

CONFIG_STAGING=y
CONFIG_STAGING_GASKET_FRAMEWORK=m
CONFIG_STAGING_APEX_DRIVER=m

- kernel version
  - 4.19.294
- Driver version
  - apex 1.0
  - gasket 1.1.2
- os
  - linux

Thanks.

reference
[1] https://coral.ai/docs/m2/get-started/#requirements

[2] https://coral.ai/docs/notes/build-coral/#required-components

[3] https://coral.googlesource.com/linux-imx/+/refs/heads/dkms/drivers/staging/gasket/

[4] https://github.com/google-coral/libedgetpu#building

### Issue Type

Bug

### Operating System

Linux

### Coral Device

M.2 Accelerator A+E

### Programming Language

Python 3.7

### Relevant Log Output

```shell
python3 examples/classify_image.py --model test_data/mobilenet_v2_1.0_224_inat_bird_quant_edgetpu.tflite --labels test_data/inat_bird_labels.txt --input test_data/parrot.jpg

I tflite/edgetpu_manager_direct.cc:453] No matching device is already opened for shared ownership.
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x1a6e, product:0x89a
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[2]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[0]
I driver/usb/local_usb_device.cc:944] EnumerateDevices: vendor:0x18d1, product:0x9302
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[2] port[0]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[2]
I driver/usb/local_usb_device.cc:979] EnumerateDevices: checking bus[1] port[0]
I tflite/edgetpu_context_direct.cc:106] USB always DFU: False (default)
I tflite/edgetpu_context_direct.cc:128] USB bulk-in queue capacity: default
I tflite/edgetpu_context_direct.cc:67] Performance expectation: Max (default)
I ./driver/mmio/host_queue.h:266] Starting in normal mode
I driver/kernel/kernel_registers.cc:83] Opening /dev/apex_0. read_only=0
I driver/kernel/kernel_registers.cc:97] mmap_offset=0x0000000000040000, mmap_size=4096
I tflite/edgetpu_context_direct.cc:401] Failed to open device [Apex (PCIe)] at [/dev/apex_0]: Internal: Could not mmap: Operation not permitted
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 160, in load_delegate
    delegate = Delegate(library, options)
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 119, in __init__
    raise ValueError(capture.message)
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples/classify_image.py", line 123, in <module>
    main()
  File "examples/classify_image.py", line 73, in main
    interpreter = make_interpreter(*args.model.split('@'))
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 87, in make_interpreter
    delegates = [load_edgetpu_delegate({'device': device} if device else {})]
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 52, in load_edgetpu_delegate
    return tflite.load_delegate(_EDGETPU_SHARED_LIB, options or {})
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 163, in load_delegate
    library, str(e)))
ValueError: Failed to load delegate from libedgetpu.so.1
```

gareth-stoyle, 2025-02-09:
This may not work but I fixed it by specifying the full path to the delegate: 

`sudo find / -name libedgetpu.so.1`

and then: 

`experimental_delegates=[tflite_runtime.load_delegate("/usr/lib/aarch64-linux-gnu/libedgetpu.so.1")`

Valeriy-Burlaka, 2025-02-28:
@GaryHuang-ASUS , we're likely on a different platforms (I'm on RPi 5), but maybe I can help 🤷

Few things I would check to deeper diagnose the issue:

1. Are you a member of the `apex` group? Did you run the following from the Coral install guide?

```sh
sudo sh -c "echo 'SUBSYSTEM==\"apex\", MODE=\"0660\", GROUP=\"apex\"' >> /etc/udev/rules.d/65-apex.rules"

sudo groupadd apex

sudo adduser $USER apex
```

2. Do you see any errors in the kernel logs related to gasket/apex?

```sh
sudo dmesg | grep -E "apex|gasket"
```

3. Do you see anything suspicious in the PCIe configuration for your accelerator modules? Pay attention to the device "Capabilities". For example, does it have `MSI-X` interrupts enabled? `MSI-X: Enable-` may be a sign of a deeper problem with initializing hardware interrupts needed by the accelerator.

```sh
sudo lspci -vv -s 03:00.0

0000:03:00.0 System peripheral: Global Unichip Corp. Coral Edge TPU (prog-if ff)
        Subsystem: Global Unichip Corp. Coral Edge TPU
...

Capabilities: [d0] MSI-X: Enable+ Count=128 Masked-
                Vector table: BAR=2 offset=00046800
                PBA: BAR=2 offset=00046068
```

Thanks!

nacabaro, 2025-05-15:
Hello, I'm also on Raspberry Pi 3b trying to run it and it's segfaulting when loading delegate. Currently I have the latest version of libedgetpu and i'm using ai_edge_litert. It works fine on my laptop (x86_64) but does not work on the raspberry pi with aarch64.